blueman-project/blueman #620: UnicodeDecodeErrors since a9357f30fded4f5934d99e07e23f19d74da18952

infirit, 2016-11-26:
I have not run python 3 for a while (to make sure we do not break 2.7) but a9357f30fded4f5934d99e07e23f19d74da18952 introduced UnicodeDecodeError under python 3 but it is not immediately obvious why. They show up for me when searching for devices just when discovery is stopped. Relevant bits from the logs below.
blueman-applet:

```
on_adapter_property_changed (/usr/lib64/python3.5/site-packages/blueman/plugins/applet/DiscvManager.py:89)
prop Discovering True 
_________
do_g_properties_changed (/usr/lib64/python3.5/site-packages/blueman/bluez/Base.py:81)
/org/bluez/hci0 Discovering True 
_________
do_g_properties_changed (/usr/lib64/python3.5/site-packages/blueman/bluez/Base.py:81)
/org/bluez/hci0/dev_7C_2F_80_3B_AF_AB RSSI -79 
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xb8 in position 1: invalid start byte
_________
on_adapter_property_changed (/usr/lib64/python3.5/site-packages/blueman/plugins/applet/DiscvManager.py:89)
prop Discovering False 

```

blueman-manager:
```
_________
do_g_properties_changed (/usr/lib64/python3.5/site-packages/blueman/bluez/Base.py:81)
/org/bluez/hci0 Discovering True 
_________
do_g_properties_changed (/usr/lib64/python3.5/site-packages/blueman/bluez/Base.py:81)
/org/bluez/hci0/dev_20_D4_E4_F6_C1_D6 RSSI -77 
_________
row_update_event (/usr/lib64/python3.5/site-packages/blueman/gui/manager/ManagerDeviceList.py:269)
row update event RSSI -77 
opacity 159
_________
do_g_properties_changed (/usr/lib64/python3.5/site-packages/blueman/bluez/Base.py:81)
/org/bluez/hci0/dev_D0_03_4B_E2_83_97 RSSI -78 
_________
row_update_event (/usr/lib64/python3.5/site-packages/blueman/gui/manager/ManagerDeviceList.py:269)
row update event RSSI -78 
opacity 156
_________
do_g_properties_changed (/usr/lib64/python3.5/site-packages/blueman/bluez/Base.py:81)
/org/bluez/hci0/dev_7C_2F_80_3B_AF_AB RSSI -79 
_________
row_update_event (/usr/lib64/python3.5/site-packages/blueman/gui/manager/ManagerDeviceList.py:269)
row update event RSSI -79 
opacity 153
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xa8 in position 0: invalid start byte
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xa8 in position 0: invalid start byte
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xa8 in position 0: invalid start byte
_________
do_g_properties_changed (/usr/lib64/python3.5/site-packages/blueman/bluez/Base.py:81)
/org/bluez/hci0 Discovering False 
```

infirit, 2016-11-28:
From my testing it seems that it always happens after ``RSSI`` changes. The annoying thing is I get no traceback so it is difficult to find it 😢

infirit, 2016-12-10:
I see RSSI set to ``nil`` in ``bluetoothctl`` and the number of lines with ``nill`` corresponds with the number of errors. It also makes sense it happens after discovery as RSSI is only available during discovery phase and at the end they get "nilled".

infirit, 2016-12-11:
Easily reproducable with below. I am now confident that it is not something we are doing wrong. But I am not sure how to proceed now, should we just ignore (it does not cause us issues afaics) or report it to the pygi folks?

```python
from gi.repository import Gio, GLib, GObject

class MyProxy(Gio.DBusProxy):
    __gsignals__ = {str('property-changed'): (GObject.SignalFlags.NO_HOOKS, None,
                     (GObject.TYPE_PYOBJECT, GObject.TYPE_PYOBJECT, GObject.TYPE_PYOBJECT))
    }

    def __init__(self, interface_name, obj_path, *args, **kwargs):
        super(MyProxy, self).__init__(
            g_name="org.bluez",
            g_interface_name=interface_name,
            g_object_path=obj_path,
            g_bus_type=Gio.BusType.SYSTEM,
            g_flags=Gio.DBusProxyFlags.NONE,
            *args, **kwargs)

        self.init()

    def do_g_properties_changed(self, changed_properties, _invalidated_properties):
        for key, value in changed_properties.unpack().items():
            path = self.get_object_path()
            print("*%s*" % self.__class__, path, key, value)
            self.emit("property-changed", key, value, path)

def print_changed_property(proxy, key, value, path):
    print("*connected function*", path, key, value)

proxy = MyProxy("org.bluez.Device1", "/org/bluez/hci0/dev_7C_2F_80_3B_AF_AB")
proxy.connect("property-changed", print_changed_property)

loop = GLib.MainLoop()
loop.run()
```

infirit, 2016-12-13:
AFAICS the problem is in the virtual function, specifically the invalidated properties. I can work around it by use GET_INVALIDATED_PROPERTIES flags instead of NONE.

For reference:
https://mail.gnome.org/archives/python-hackers-list/2016-December/msg00003.html
https://github.com/blueman-project/blueman/pull/628

infirit, 2016-12-27:
Found a bug open on it https://bugzilla.gnome.org/show_bug.cgi?id=756009

cschramm, 2017-01-03:
So I guess we can close this as it's not our fault and we have a workaround.

infirit, 2017-01-03:
Yup, closing now that I merged the workaround,

closed 2017-01-03 by infirit