blueman-project/blueman #624: No devices is detected

escapeplan4, 2016-12-05:
I've got linux mint sarah 18 and although bluetooth is on it can't find any devices.

infirit, 2016-12-08:
Which version of blueman does mint 18 come with? 

Does ``bluetoothctl`` show any device? If so please post the log of blueman-applet as described in https://github.com/blueman-project/blueman/wiki/Troubleshooting. If not there is probably something not right with bluetooth in general on the machine.

escapeplan4, 2016-12-15:
> project/blueman/wiki/Troubleshooting. If not there is probably something

no one, i've try this because default bluberry fail also in mint 18

infirit, 2016-12-16:
>no one, i've try this because default bluberry fail also in mint 18 

Blueberry and blueman both use the same bluetooth stack called bluez. If it does not work in both blueberry and blueman that indicates there is a general problem and it is not specifically related to blueman.

But I'll try yo see if we can make it work.
 * What is your device? run ``bluetoothctl list`` from a terminal
 * Does you device have hardware kill switch (sometimes called airplane mode for wireless radios)

infirit, 2017-01-10:
No response for a month, closing. Feel free to comment with the info and we can re-open.

closed 2017-01-10 by infirit