blueman-project/blueman #634: Multiple file send from phone only receives one file

Rudd-O, 2016-12-25:
Master branch when I send to my computer multiple files, only the first is received and notified for.

The version available on Fedora does not have that problem.  This is a new bug.

infirit, 2016-12-25:
We dropped bluez 4 support in master which may have caused this. I am not at my machine for the next several days but if you are interested the code for this lives in TransferService if I am not mistaken.

Rudd-O, 2016-12-26:
I am famliar with the code but after toying with it, I couldn't get it
to work.  Sorry, but the code is broken, and whether it was caused or
not by Bluez4 being dropped, it needs to be fixed.

infirit, 2016-12-26:
No worries, I may have found the problem. I'll confirm tomorrow and if correct I'll push a fix.
-- 
Sent from my Android device with K-9 Mail. Please excuse my brevity.

infirit, 2016-12-26:
Silly bug introduced recently, https://github.com/infirit/blueman/commit/8c013b35401f28868cdfd78d8854ab0c844f60c8 should fix but not properly tested yet. Will do and push tomorrow.

infirit, 2016-12-26:
Transferred 38 files with no problems, pushed as 8c013b35401f28868cdfd78d8854ab0c844f60c8.

closed 2016-12-26 by infirit