blueman-project/blueman #639: Error message on every startup

nyhudson, 2017-01-07:
I get the following error message on every startup:

"Configured directory for incoming files does not exist"

I try to open blueman-services and there are no options to change this. Actually, when I open blueman-services I get a lot of python error messages. This is really annoying behavior and I would suggest you make your software a little more user friendly!

Am using Fedora 25. Anyway, will uninstall this software.

infirit, 2017-01-07:
Unless something changed blueman is still broken in fedora, see https://github.com/blueman-project/blueman/issues/537 for some background. Closing as a dupe of #537.

closed 2017-01-07 by infirit

nyhudson, 2017-01-08:
Thanks for the reply. Well, then Fedora really shouldn't include blueman until it is fixed. I should go yell at Fedora devs instead :-)