blueman-project/blueman #640: TransferService is not loadable (gives huge error message)

Vednier, 2017-01-09:
Hello, just today i encountered exceptionally weird bug - i noticed that Blueman cannot receive files sent from mobile device, checked if something missing and found that TransferService is not enabled, i tried to enable it and...got this. I'm sorry it in image file but i dont see any way to copy-paste this.
[image]
Gettin this error windows every time i trying to enable TransferService.
Using Xubuntu 16.10 64 bit(XFCE), Blueman 2.0.4, Bluez 5.41, Bluez-obexd 5.41 is also present.
So far i really dont know that to add.

infirit, 2017-01-09:
This means that the obex daemon is not enabled.

As a user do ``systemctl --user start obex`` and see if the error goes away. If it does you can enable it globally with ``sudo systemctl --global enable obex``.

Vednier, 2017-01-11:
Yes, thats worked. I had no such issue previously, on 16.04 so didnt expected that. thanks.

closed 2017-01-11 by Vednier

infirit, 2017-01-11:
Its the introduction of systemd you are suffering from. The same is seen on all systemd distros including Fedora and Arch.