blueman-project/blueman #646: Need icon for portable audio device

infirit, 2017-01-20:
I now have a small BT loudspeaker which shows in blueman-manager as the blueman icon. Based on its class it would  use "blueman-portable-audio.png" but we have none.

I don't know who made the icons bur my artistic skills are at the pre-school level. Maybe there is an artists lurking what is willing to help?

cschramm, 2017-03-28:
Not sure what the spec means with "Portable Audio". If it's always loudspeakers we could just reuse blueman-loudspeaker but I guess it could also e.g. be some media player device, so finding an icon that generically makes sense is hard...

infirit, 2017-03-28:
Commit a7c170e5abee3702025792e8e4905b6e4e074ee1 now adds the fallback so the urgency is not that high.

I do not know what the spec says but the range of form factors these speaker come in is very diverse so using ``blueman-loudspeaker`` would be fine.

infirit, 2019-10-11:
We removed these icons so this can be closed now.

closed 2019-10-11 by infirit