blueman-project/blueman #648: Logitech MX Anywhere 2 UI glitches

akostadinov, 2017-01-22:
Hello, I've got Logitech MX Anywhere 2 mouse to hopefully find a good mouse finally. It connects perfectly well so far which is beyond awesome (unlike the lenovo thinkpad bluetooth keyboard which seems to be doing legacy pairing, connects unreliably and is laggy /btw can I somehow force disable legacy pairing?/).

issues I see:

- blueman applet doesn't show a mouse icon
- blueman applet does not show connection strength indicators
- battery power level is not shown through the power manager applet (for the lenovo keyboard it was shown)
- a number of unknown UUIDs in the `info` section

Here is a screenshot: http://pasteboard.co/p8ewixAgO.png

I suspect you will tell me that some of these issues are outside control of blueman. I'll move any of these to other projects if you advise so.

infirit, 2017-01-22:
>blueman applet doesn't show a mouse icon

It uses the ``Class`` property for this which is empty for you. I think we may want to use the ``Icon`` property as a fallback. @cschramm, opinions, sugestions? Drop icons based on Class entirely and solely use ``Icon``?

>blueman applet does not show connection strength indicators

Can you post the console output from ``blueman-manager`` it should have useful info.

>battery power level is not shown through the power manager applet (for the lenovo keyboard it was shown)

That is out of our control, probably best to report this to the people from the power manager applet.

> a number of unknown UUIDs in the info section

These are LE uuids not part of SDP from what I could quickly find. We will need to handle this at some point but as long as it does not crash I'm ok with this for now. Also reff  #625

akostadinov, 2017-01-22:
It is very interesting. The icon of blueman applet shows green as if device is connected. But if I right click on manager over the mouse, it doesn't show the option to disconnect as if device is not connected. But the mouse is in fact connected and working. If I turn it off through button, then blueman applet icon shows as if no devices are connected. So to me it seems that the manager app doesn't recognize the mouse is connected presently. Also doesn't show any "Connect to" options with right-click. 

This log was taken like:

1. turn off mouse
2. start manager in console
3. turn on mouse
4. right click on mouse in manager

The mouse just starts to work so I didn't notice manager didn't show "Connect" option. For some reason I can't take a screenshot though. It is just "Info" at the top, then grayed out "Send FIle", etc.

[manager.txt](https://github.com/blueman-project/blueman/files/722124/manager.txt)

cschramm, 2017-02-02:
I think class vs icon is actually done inconsistently as it seems blueman-assistant *does* use the Icon property.

Missing UUID names are added in #656.

The missing connect / disconnect is caused by blueman not knowing any of the device's services. It could be fixed with #575 *plus* the generic connect / disconnect needs to be shown whenever GATT services (UUIDs with a first byte of 0x18) are present.

infirit, 2017-02-02:
cool, i'll update the generic (dis)connect PR later today.
-- 
Sent from my Android device with K-9 Mail. Please excuse my brevity.

akostadinov, 2017-02-02:
I can test the changes. Just tell me commit id when ready (can be before merging to master).

btw unrelated note, with kernel 4.8 mouse was behaving very well. I upgraded to 4.9 and it started to lag.

infirit, 2017-02-03:
@akostadinov the uuid names have been updated in master as is the weird icon problem.

@cschramm, Yes, it is only used in blueman-managers devicelist. The advantage is that we have a consistent icon style in blueman-manager. For now I can just add a fallback to use the Icon property, if you disagree let me know :)

infirit, 2017-02-03:
@akostadinov also, when the mouse is connected. Can you try to read the rssi, req and lq with hcitool.

```
hcitool rssi <mouse mac>
hcitool lq <mouse mac>
hcitool tpl <mouse mac>
```

@cschramm, it looks like we fail to initialize the connection which raises and exception and we end up completely dropping the bars. The code in ManagerDeviceList can handle failing calls to the conn_info getters so something like below would at least show the bars. Not sure if this is a wise thing to do though..

```diff
diff --git a/module/_blueman.pyx.in b/module/_blueman.pyx.in
index b10bfff0..e85b7971 100644
--- a/module/_blueman.pyx.in
+++ b/module/_blueman.pyx.in
@@ -273,10 +273,13 @@ cdef class conn_info:
         self.hci = int(hci_name[3:])
         res = connection_init(self.hci, addr, &self.ci)
         if res < 0:
-            raise ConnInfoReadError(ERR[res])
+            self.is_connected = False
+        else:
+            self.is_connected = True
 
     def deinit(self):
-        connection_close(&self.ci)
+        if self.is_connected:
+            connection_close(&self.ci)
 
     def get_rssi(self):
         cdef char rssi
```

akostadinov, 2017-02-04:
```
$ hcitool rssi MAC
Get connection info failed: No such file or directory
```

This works with other devices but not with the mouse. All 3 command have same result. I read now this is expected for LE devices: http://stackoverflow.com/questions/24417017/bluetooth-le-signal-strength-linux

I tried the suggested `hcitool lescan` + `btmon` but still could not obtain connection quality. In any case, would be good to have at least connect/disconnect option if quality cannot be assessed. But will be awesome if connection quality is shown 😀 

```
@ RAW Open: hcitool (privileged) version 2.22          {0x0002} [hci0] 0.127143
@ MGMT Open: bluetoothd (privileged) version 1.14             {0x0001} 0.127144
@ MGMT Open: btmon (privileged) version 1.14                  {0x0003} 0.127169
> ACL Data RX: Handle 3585 flags 0x02 dlen 14                   [hci0] 1.001384
      ATT: Handle Value Notification (0x1b) len 9
        Handle: 0x0031
          Data: 0000ff0f000000
... many more similar messages to the last one for every mouse move
```

infirit, 2017-02-04:
>This works with other devices but not with the mouse. All 3 command have same result. I read now this is expected for LE devices

I was afraid of that but better to be sure. I don't have proper LE devices so not sure how much I can work on this.

cschramm, 2017-02-07:
If that's the way to get the RSSI we pretty much depend on BlueZ to at least process the LE Meta Event and provide the captured value via its the dbus API, which would make most sense via the RSSI property we're already using. So, unfortunately, I do not see much we could do about it.

I'll skim through the BlueZ source when I find some time to see what's currently going on there.

Interestingly I was not able to find the RSSI in `btmon` as described in the Stack Overflow thread (using BlueZ 5.43). 🤔

cschramm, 2017-02-08:
So. BlueZ' handling is pretty straightforward. The value exposed as the RSSI property is simply what the kernel delivers in `MGMT_EV_DEVICE_FOUND` events. It seems like not even BlueZ is aware of the RSSI of LE devices. 🙁

infirit, 2017-02-09:
#658 adds a fallback that uses the ``Icon`` property from the bluez.Device.

infirit, 2017-02-09:
>The value exposed as the RSSI property

Assuming you are referring to the icon exposed by the dbus api it is not useful for our use case. The property is only used when the adapter is discovering. The moment we stop discovering the property is invalidated.

infirit, 2017-02-12:
@akostadinov would be great if you can try master and let us know how things are. The signal bars will still be missing though but at least the other issues should be solved. Thx

akostadinov, 2017-02-12:
Hi, good improvements, thank you! Here's the facts sheet:

- `disconnect` button still non-clickable
- power indicators not visible (as expected)
- `00010000-0000-1000-8000-011f2000046d Unknown` device id (btw would have been really awesome if one can copy text off the device `info` window)
- mouse icon++ looks awesome now
- applet icon shows up at all times (there is  a minor glitch - when turning bt on, then icon disappears for a second until it shows up again)

cschramm, 2017-04-28:
Can we consider this done?

akostadinov, 2017-04-28:
Well, some things as explain above are still not working. If nobody is going to work on them, it is up to you whether you close or not.

cschramm, 2017-04-28:
> `disconnect` button still non-clickable

Did you try the current code from #575?

- power indicators not visible (as expected)

I think there's nothing we can do about that. It's just a downside of LE.

> `00010000-0000-1000-8000-011f2000046d Unknown`

That's a proprietary service. Makes sense that's it's unknown. 😄

infirit, 2017-04-29:
@cschramm 
>That's a proprietary service. Makes sense that's it's unknown. 😄

Have not looked at the spec but do we have a way of identifying those proprietary ones? 

@akostadinov 
>btw would have been really awesome if one can copy text off the device info window

That should be fairly easy #684. When I have some time I'll have a look.

cschramm, 2017-04-30:
@infirit: Identifying in terms of detecting that it is a proprietary service? Yes. Pre-allocated UUIDs follow the pattern `0000XXXX-0000-1000-8000-00805F9B34FB`. `XXXX` is the equivalent 16-bit value and `XXXX0000` is the equivalent 32-bit value (no idea what that is there for).

Actually our handling of UUIDs is wrong whenever we use `uuid128_to_uuid16` without checking that the other bits match the Bluetooth Base UUID `00000000-0000-1000-8000-00805F9B34FB`. Theoretically there could be a proprietary service `0001110a-0000-1000-8000-011f2000046d` which does not have corresponding 16-bit  and 32-bit values but we would extract `110a` anyway and consider it the pre-allocated audio source service. I guess the easiest fix is to check for the Bluetooth Base UUID in `uuid128_to_uuid16` and return something like `None` or `0000` if it does not fit. I'll look into it.

If you meant identifying in terms of giving it a name, well, maybe. The SDP provides much more than just the UUIDs (ServiceID), also e.g. ProviderName, ServiceName, ServiceDescription, so it should be available. I cannot find a way to get that information from BlueZ though...

infirit, 2017-04-30:
>Identifying in terms of detecting that it is a proprietary service?

Yes and marking them as "Proprietary" instead of Unknown, no need to get the actual name.

>I guess the easiest fix is to check for the Bluetooth Base UUID in uuid128_to_uuid16 and return something like None or 0000 if it does not fit. I'll look into it.

Perhaps have a UUID object that handles all of this transparently. Something like

```python
class UUID(object):
	def __init__(self, uuid):
		self.__uuid = uuid
		self.is_proprietary = "0000-1000-8000-011f2000046d" in uuid.lower()
		self.is_base = "0000-1000-8000-00805f9b34fb" in uuid.lower()
	@property
	def uuid16(self):
		if self.is_proprietary or self.is_base:
			return int("0x" + self.__uuid[4:8], 16)
		else:
			return None
	@property
	def name(self):
		if self.is_proprietary:
			return "Proprietary"
		elif self.is_base:
			return uuid_names[self.uuid16]
```

closed 2017-11-28 by cschramm