blueman-project/blueman #650: Error when connecting devices

One-SNapp, 2017-01-26:
When i try to connect by bluetooth headphones using blueman i get the error:
Connection Failed: blueman.bluez.errors.DBusFailedError: Resource temporarily unavailable

Also it looks like it connects, but disconnects a second later. The headphones work perfectly on windows and ipad. They used to work on blueman.

I'm using ubuntu.

parkerlreed, 2017-01-26:
Do you have the pulseaudio bluetooth module installed? Might just be that it can't talk to pulse and make a connection.

lecho74, 2017-01-27:
I get the same error with a bluetooth mouse (since last update),
but with Input/output error...
blueman 2.0.4 on Linux Mint 18

One-SNapp, 2017-01-28:
I tried that. It didnt work

infirit, 2017-01-28:
Do post the output from ``blueman-applet`` and ``blueman-manager`` as described here. Also try to use ``bluetoothctl`` to connect and report back if that works or not.

One-SNapp, 2017-01-29:
Sorry, how do i do that?

infirit, 2017-01-29:
for bluetoothctl, open a terminal and run that command. Then a list of known devices will be listed. Then in blutoothctl issue the connect command with the mac of your device. The arch wiki has an example session https://wiki.archlinux.org/index.php/bluetooth#Configuration_via_the_CLI

Forgot to post the link for the logs https://github.com/blueman-project/blueman/wiki/Troubleshooting#debugging

agrill, 2017-01-29:
Same issue here, pairing to Marantz NR1506 works, but connection fails. Tried connecting with hcitool and bluetoothctl, always the same error:

root@grill-ZBook-14:/home/grill# bluetoothctl 
[NEW] Controller 80:19:34:B7:82:C9 xubuntu-0 [default]
[NEW] Device 34:BE:00:F5:B1:C6 SM-G900F
[NEW] Device F0:65:DD:97:6E:F7 HP Bluetooth Mouse X4000b
[NEW] Device 00:06:78:2A:D8:AE Marantz NR1506
[bluetooth]# connect 00:06:78:2A:D8:AE
Attempting to connect to 00:06:78:2A:D8:AE
[CHG] Device 00:06:78:2A:D8:AE Connected: yes
Failed to connect: org.bluez.Error.Failed
[CHG] Device 00:06:78:2A:D8:AE Connected: no
[CHG] Device 00:06:78:2A:D8:AE Connected: yes
[CHG] Device 00:06:78:2A:D8:AE Connected: no

Log files attached, please tell me if you need more.

[bluetoothd.txt](https://github.com/blueman-project/blueman/files/737871/bluetoothd.txt)
[hcidump.txt](https://github.com/blueman-project/blueman/files/737870/hcidump.txt)

lecho74, 2017-01-30:
Same issue here with unknown bluetooth mouse
CHG] Device 00:00:00:00:1F:09 Connected: no
[CHG] Device 00:00:00:00:1F:09 Connected: yes
[CHG] Device 00:00:00:00:1F:09 Connected: no

parkerlreed, 2017-01-30:
Just checking but be sure

````
load-module module-bluetooth-policy
````

is in your default.pa in /etc/pulse. I think I had to enable that at one point or another but should be default. Either way if that's failing in bluetoothctl then this isn't the fault of blueman and you would have to fine an issue with your distro/upstream bluez repo.

agrill, 2017-01-30:
Yes, double checked if pulseaudio bluetooth modules are loaded via paman. Blueman reports it as a DBUS error, but probably it is a problem within bluez.

Filed a bug at:
https://bugs.launchpad.net/ubuntu/+source/bluez/+bug/1660277

infirit, 2017-01-30:
Blueman talks to BlueZ over the dbus api and pairs, connects etc using that. If indeed bluetoothctl also does not work then we most likely can not fix that in blueman.

But, please post the blueman-applet log, it may contain some useful info. It may be something is powering down the adapter and then all connection and pair attempts will fail. One thing to check of the killswitch got enabled for some reason, as root run ```rfkill list``` to see if one or more are blocked.

Also all of you on ubuntu or mint?

One-SNapp, 2017-02-02:
[NEW] Controller B8:81:98:17:3F:A3 s [default]
[NEW] Device D0:8A:55:09:19:F0 Hesh 2 Wireless
[NEW] Device 18:EE:69:29:B7:C8 Apple TV
[NEW] Device E4:98:D6:AA:FA:F6 SSD-DMPMTADPF182-iPad
[NEW] Device 0C:84:DC:DA:C6:81 VBTL820-DA
 I'm trying to connect to the hesh 2 wireless

cschramm, 2017-02-03:
@agrill: You're not getting `Resource temporarily unavailable`, are you? I'm not profound in HCI commands, but your bluetoothd output shows that the device rejects the authentication and in your bluetoothctl snippet you do not enable an agent which might just be the cause for that. If blueman-applet was running, that could indicate an issue there as it should provide an agent that should also get used if you trigger a connection from within bluetoothctl, however we'd need the output to look into that then.

@lecho74: We've seen "Input/output error" e.g. in #413 where it is caused by compatibility issues with the BCM2046 adapter (which might be resolved in recent versions of Linux / BlueZ) or in #240 where a BlueZ updated fixed it (adapter and versions unknown).

@One-SNapp: What are your versions of blueman, bluez, and pulseaudio-module-bluetooth?

lecho74, 2017-02-03:
@cschramm: I have a intel AC 8260 wifi / bluetooth adapter.

There is Linux Mint 18 on my Laptop and try to stay updated (maybe a mistake...)
I am no linux experte (but willing to learn) but such problems takes more knowlegde
and "scripting" than i know this time...

BTW: My laptop is a Acer Aspire ES1-131-P66Z with SSD, 8GB Ram & Intel AC8260.

agrill, 2017-02-04:
@cschramm: Indeed, I get Resource temporarily unavailable. I also noticed the rejected connection from the remote device, but thought maybe a handshake problem.
What kind of logs would you need exactly to track down the problem?

cschramm, 2017-02-07:
@lecho74: From a quick google search it looks like AC 8260 can be a very weird device as well. `dmesg` might give a hint what's going on.

@agrill: https://github.com/blueman-project/blueman/wiki/Troubleshooting#debugging

lecho74, 2017-02-07:
@cschramm: weird, that is it indeed. After some activating / deactivating / diagnosis (dont ask me what i did :) the adapter works prpoperly again - no one knows how long.

Anyway thanks to all - You are doing great work!

cschramm, 2017-02-07:
@lecho74: Funniest thing I read on the Internets is that the adapter does not work at all anymore after booting into Windows until you reset your CMOS. 😂 https://bbs.archlinux.org/viewtopic.php?id=218895

lecho74, 2017-02-07:
> just like windows kidnapping my bluetooth adapter

That is why i try to life without windows (it is not easy - for me - but i try)

parkerlreed, 2017-02-07:
@cschramm I remember back when WUBI was a thing and disabling a device in Windows made it completely disappear for the Ubuntu install. Kinda funny.

One-SNapp, 2017-02-09:
pulseaudio 8.0 Not sure how to find the versions for bluez, and pulseaudio-module-bluetooth

cschramm, 2017-02-12:
`dpkg -l bluez` and `dpkg -l pulseaudio-module-bluetooth`

One-SNapp, 2017-02-12:
bluez: 5.37 
I get 1:8.0-0 for pulseaudio module bluetooth

agrill, 2017-02-17:
@infirit @cschramm Sorry for the delay, just came back to this. Attached some more logs, the blueman-applet log is when trying to connect via GUI.
I'm on Xubuntu 16.10.
Please tell me if I can further help to track down the problem.

[dpkg.txt](https://github.com/blueman-project/blueman/files/781872/dpkg.txt)
[journalctl.txt](https://github.com/blueman-project/blueman/files/781873/journalctl.txt)
[blueman-applet.txt](https://github.com/blueman-project/blueman/files/781874/blueman-applet.txt)
[lournalctl.txt](https://github.com/blueman-project/blueman/files/781870/lournalctl.txt)
[rfkill.txt](https://github.com/blueman-project/blueman/files/781871/rfkill.txt)
[pulseaudio.txt](https://github.com/blueman-project/blueman/files/781869/pulseaudio.txt)
[bluetoothctl.txt](https://github.com/blueman-project/blueman/files/781875/bluetoothctl.txt)

agrill, 2017-02-17:
Oh, forgot to mention: I read the manual from the reciever, it says the bluetooth adapter is A2DP only, my Android phone connects fine.

cschramm, 2017-04-28:
@agrill: The `connect error: Permission denied (13)` in the syslog yields all kinds of issues on Google and I guess there's not much more to get out of BlueZ as it's just the receiver rejecting the connection attempt. Maybe it's not really paired (paired to that Android phone instead?) and needs some reset, see e.g. https://bbs.archlinux.org/viewtopic.php?id=204875. I cannot find any reference to Marantz devices though.

agrill, 2017-09-10:
@cschramm: I updated to ubuntu 17.04 today and finally was able to connect without any difficulties. Still don't know if it was a blueman or bluez problem, but seems to be resolved. Thanks for your effort!

closed 2017-09-11 by cschramm

nekromoff, 2017-10-25:
Still having this error in Ubuntu 17.10

After trying to connect I get Resource temporarily unavailable, while blueman seems to be connected for about two or three seconds to the headset.

cschramm, 2017-10-27:
@nekromoff

I've seen `Resource temporarily unavailable` caused by some bluetooth devices and re-pairing, i.e. removing the paired device on the Linux side and then pair both as usual might help. Did you try that?

To get more information on the issue, you can stop the bluetooth service and start bluetoothd manually with the `-d` and `-n` flags.

agrill, 2017-10-27:
When I had to setup a fresh 17.10 install lately, the problems reappeared. After some more debugging, I finally found out that the problems were caused by an "Always on" option in the network setting of the receiver.
This causes the network hardware to be powered on even in standby (for automatic power on). For some reason the receiver seems to handle the connection attempts (and terminations) and/or pairing improperly then. 
Long story short, after disabling this feature, everything works properly without any issues on a fresh install without any tweaks.

e-orz, 2017-11-13:
@agrill I have ubuntu 17.10 with the same issue. How do I disable the "Always on" option?
Thanks!

nekromoff, 2017-11-13:
I force restart `/etc/init.d/bluetooth restart` and this way it works
again. Shouldn't be necessary, regular user will not be able to fix this.

infirit, 2017-11-14:
Not sure what is meant with always on but we have ``AutoEnable`` in ``/etc/bluetooth/main.conf``.
<pre># AutoEnable defines option to enable all controllers when they are found.
# This includes adapters present on start as well as adapters that are plugged
# in later on. Defaults to 'false'.
#AutoEnable=false</pre>

Or there is in blueman powermanager plugin which will attempt to power it on.

agrill, 2017-11-14:
@e-orz There is a "Network-ctrl" setting in my Marantz receiver under Setup -> Network. I had to put it from "Always on" to "Off (in Standby)" (Names could be slightly different since I have a German menu). Since then everything works fine. 

@infirit The receiver mentioned in my last post was my AV receiver, not my laptop.

thomasphansen, 2017-11-23:
> Funniest thing I read on the Internets is that the adapter does not work at all anymore after booting into Windows until you reset your CMOS.  https://bbs.archlinux.org/viewtopic.php?id=218895
--

@cschramm and all: just some useful information in case someone hits the same problem I was in:

I have the intel 8087:0a2a wifi/bluetooth hybrid adapter. Bluetoot was working perfectly, until i had the _brilliant_ idea of enabling the bluetooth on a Windows 10 VirtualBox VM. After that, bluetooth would never connect again on linux, even after I completely removed it from the VM. The error would be:

    Connection Failed: blueman.bluez.errors.DBusFailedError: Resource temporarily unavailable

I tried EVERYTHING, including upgrading kernel, bios, messing around with all possible settings. And it turned out that @cschramm tip (resetting the CMOS) was the only working solution.

So, BEWARE: Windows VMs can screw your bluetooth just as much as a dual-boot configuration, and the only way - AFAIK - to fix the mess is to clean CMOS...

Thanks, man! :)

JimMF, 2017-12-05:
I ran into a similar problem with a usb bluetooth adaptor that uses the Broadcom BCM20702 chipset. I had to load an updated firmware for that chipset to resolve it. If you know your usb adaptor uses this chipset you might want to give it a try. Here's a link to the adaptors website page that provides the firmware update with installation instructions. I hope this helps. https://plugable.com/2014/06/23/plugable-usb-bluetooth-adapter-solving-hfphsp-profile-issues-on-linux/

x-ji, 2018-02-10:
Also had this issue with my MDR-1000X headphone. The status changes between "Connected: yes" and "Connected: no" all the time. Had to unpair and then re-pair the device every time for it to finally connect.

Cyberxine, 2018-04-05:
I get the 'Connection Failed: blueman.bluez.errors.DBusFailedError: Resource temporarily unavailable' message when attempt to connect to a previously paired set of BlueTooth headphones/headset after using them in Windows.
I tried using bluetoothctl to connect to the device and it seemed to connect and immediately disconnect, 'Yes'-'No'-'Yes'-'No', then throws the DBus error message.
I found that by using the tray applet (I'm using Ubuntu 17.10 MATE) to turn Bluetooth off/on then re-pairing was a viable solution for the issue.

amartin1911, 2018-05-03:
Got the same error of "connected" and then "disconnected" immediately.
@nekromoff solution worked for me on Ubuntu 18 (Bionic Beaver).

telpesoron, 2018-05-09:
Some issue of connected /disconnected /connected / disconnected here, using Sony MDR headphones.
The most quick way for me is restarting bluetooth service with

`sudo service bluetooth restart`

and connecting to headphones through audio sink.

Dagefoerde, 2018-05-18:
I agree with @amartin1911, this problem is still relevant in Ubuntu 18.04 Bionic. The workaround of @telpesoron and @nekromoff works fine.

gaborcsapo, 2018-06-11:
+1 also have this problem in Ubuntu 18.04 with my JBL bluetooth speaker.

wheelerlaw, 2018-06-13:
Similar to @JimMF's issue with the Broadcom BCM20702 chipset, if you run `dmesg | grep -i bluetooth` and see something that resembles:

```
[    4.752206] bluetooth hci0: Direct firmware load for brcm/BCM20702A1-0a5c-21e6.hcd failed with error -2
[    4.752207] Bluetooth: hci0: BCM: Patch brcm/BCM20702A1-0a5c-21e6.hcd not found
```

... my solution was to download the appropriate firmware image from [winterheart/broadcom-bt-firmware](https://github.com/winterheart/broadcom-bt-firmware/tree/master/brcm) and place it in your `/lib/firmware/brcm` directory.

Cyberxine, 2018-06-14:
Wheeler's solution worked perfectly. After restarting my machine I had a
single error from dbus stating 'In progress' and I figured I I'd just try
reconnecting.
Though, I did delete the device before cold-booting and repairing (where I
saw the 'in progress' error). Someone else should try the solution without
unpairing the device to see if it works in that case as well.

filipuic, 2018-06-29:
`service bluetooth restart` worked for me.

fcuencamargalef, 2018-07-22:
Same issue in 18.04 with Bose Speakers.
service bluetooth restart works fine.

jessewriter, 2019-07-02:
I had this error message from my Linux Mint 19.2 and my Amazon Echo Show.  Tried everything then gave up and rebooted the Echo Show... should have tried the magic reboot first fixes so many problems... lol

hildakh, 2020-04-07:
> `service bluetooth restart` worked for me.

Worked for me too! Thanks! I had this issues for days!

sergio-ivanuzzo, 2021-01-04:
For guys, who tried to use bluetooth devices with dual boot: did you tried to share pair keys between OSes ? https://unix.stackexchange.com/a/255510/159027