blueman-project/blueman #652: Blueman now asking for Polkit authentication twice

posophe, 2017-01-30:
Quoted from https://bugzilla.suse.com/show_bug.cgi?id=1006601
"Hello, with the Cinnamon desktop (and probably others) the Blueman applet now causes two ugly authentication prompts immediately after login. It is missing PolKit authentication for: org.blueman.network.setup and org.blueman.rfkill.setstate ."

It actually prevents Blueman to be officially included in openSUSE repos, and besides that, to become our official bluetooth client

infirit, 2017-01-30:
Yes, see below from the PR  #506. Please make sure to change the group so it complies with OpenSuse rules. On gentoo the ``plugdev`` group is used which is the same one for NetworkManager.

If you have further problems let us know.

>@posophe this is for #423
>
>For all, I don't have a machine with a hardware killswitch (only have usb dongles) so I really would appreciate if people can make sure this works properly.
>
>To not have your users to authenticate all the time distros typically ship with a polkit rules file. See below for an example.
>
>````
>/* Allow users in wheel group to run blueman network config without authentication */
>polkit.addRule(function(action, subject) {
>    if ((action.id == "org.blueman.network.setup" ||
>         action.id == "org.blueman.dhcp.client" ||
>         action.id == "org.blueman.rfkill.setstate" ||
>         action.id == "org.blueman.pppd.pppconnect") &&
>        subject.isInGroup("wheel")) {
>        return polkit.Result.YES;
>    }
>});
>````

infirit, 2017-01-30:
btw @posophe, from [comment 5 pm bug 987141](https://bugzilla.suse.com/show_bug.cgi?id=987141#c5)

>For instance, in DisconnectDevice(), disconnect_service(), SetPluginConfig()
and many others, its missing.

All the privileged operations are done by ``blueman-mechanism`` and it runs are root. The ones mentioned are normal dbus calls done on ``blueman-applet`` that runs as the normal user. Why would those need authentication?

posophe, 2017-01-30:
No idea. I've just forwarded this new opened issue. But I'll forward your comments on our bugzilla tracker.

infirit, 2017-01-30:
I subscribed to the two issues and reply if needed ;-)

cschramm, 2017-02-03:
Closing as I guess there's nothing to do on our side at the moment. Feel free to reopen if any further questions or issues arise.

closed 2017-02-03 by cschramm

infirit, 2017-02-06:
For completeness, the review was completed: https://bugzilla.suse.com/show_bug.cgi?id=987141#c14