blueman-project/blueman #653: do not hide applet icon when bluetooth is soft disabled

akostadinov, 2017-01-31:
Hi, there is a great usability issue here. If I right click on applet icon and choose "turn off", then the icon disappears and there is no way to re-enable bluetooth. I had to install the `rfkill` cli tool to get bluetooth back.

I think by default icon should not disappear when bluetooth is only off by software. Can be user configurable but I can't imagine anybody using bluetooth regularly to desire it invisible.

parkerlreed, 2017-01-31:
The icon _shouldn't_ disappear. When I right click and turn off on my system I get a little Bluetooth icon with a red x on it. Could it just be that your icon theme doesn't include an icon for it?

akostadinov, 2017-01-31:
@parkerlreed I don't know, what is the name/filename? I just have a normal install on Fedora.

parkerlreed, 2017-01-31:
bluetooth-disabled.png is the name of it. Oddly enough the icon I see when disabled isn't in my /usr/share/icons/ at all from what I can see.

akostadinov, 2017-01-31:
I have that icon. Strange. It must be something else.
/usr/share/icons/hicolor/48x48/status/bluetooth-disabled.png

parkerlreed, 2017-01-31:
Try running the applet from your terminal and see if it prints any messages when you do the right click > turn off. Might be an oddity with the hardware/software and it may be quitting completely.

parkerlreed, 2017-01-31:
Odd I just tested with mine and the icon disappears AFTER I tell it to turn back on.

Here's the applet output http://ix.io/1RCr

infirit, 2017-01-31:
@akostadinov which version are you using and please also post the log of blueman-applet.

I broke my bluetooth so I can't check right now, marking as a bug until I can look at it.

parkerlreed, 2017-01-31:
On mine I'm using blueman-git 2.1.alpha1.r58.gc99da003-1

infirit, 2017-02-01:
Master works fine for me, it does however have an issue that the icon does not disappear when the adapter is removed. But the icon changes properly when disconnected by adding a red cross.

akostadinov, 2017-02-01:
Using "blueman-2.1-0.7.gitf810c73.fc24.x86_64". Perhaps I have to rebuild with HEAD.

akostadinov, 2017-02-01:
Just installed 299b5a6 which shows up as blueman: 2.1.alpha1.

It doesn't hide icon. But when I re-enable bluetooth, the icon disappeared. I see in console:

```
Traceback (most recent call last):
  File "/usr/lib/python3.5/site-packages/blueman/gui/manager/ManagerDeviceList.p
y", line 202, in on_finished
    super(ManagerDeviceList, self).device_remove_event(device, tree_iter)
  File "/usr/lib/python3.5/site-packages/blueman/gui/DeviceList.py", line 206, i
n device_remove_event
    sig = self.__signals.pop(device.get_object_path())
KeyError: '/org/bluez/hci0/dev_CC_B1_1A_DA_CE_47'
Traceback (most recent call last):
  File "/usr/lib/python3.5/site-packages/blueman/gui/manager/ManagerDeviceList.p
y", line 202, in on_finished
    super(ManagerDeviceList, self).device_remove_event(device, tree_iter)
  File "/usr/lib/python3.5/site-packages/blueman/gui/DeviceList.py", line 206, i
n device_remove_event
    sig = self.__signals.pop(device.get_object_path())
KeyError: '/org/bluez/hci0/dev_0C_B3_19_ED_1F_B3'
Traceback (most recent call last):
  File "/usr/lib/python3.5/site-packages/blueman/gui/manager/ManagerDeviceList.py", line 202, in on_finished
    super(ManagerDeviceList, self).device_remove_event(device, tree_iter)
  File "/usr/lib/python3.5/site-packages/blueman/gui/DeviceList.py", line 206, in device_remove_event
    sig = self.__signals.pop(device.get_object_path())
KeyError: '/org/bluez/hci0/dev_E7_A4_DF_BF_F5_FB'
Traceback (most recent call last):
  File "/usr/lib/python3.5/site-packages/blueman/gui/manager/ManagerDeviceList.py", line 202, in on_finished
    super(ManagerDeviceList, self).device_remove_event(device, tree_iter)
  File "/usr/lib/python3.5/site-packages/blueman/gui/DeviceList.py", line 206, in device_remove_event
    sig = self.__signals.pop(device.get_object_path())
KeyError: '/org/bluez/hci0/dev_90_7F_61_10_E6_39'
/usr/lib64/python3.5/site-packages/gi/overrides/GObject.py:541: Warning: gsignal.c:2635: instance '0x55c2a721d270' has no handler with id '6976'
  return func(*args, **kwargs)
```

I also see this but probably unrelated:
```
(wrapper-1.0:1863): Gtk-CRITICAL **: IA__gtk_window_resize: assertion 'GTK_IS_WI
NDOW (window)' failed
Fontconfig warning: ignoring C.utf8: not a valid language tag
```

It reproduces consistently for me.

infirit, 2017-02-01:
Yes, there is something weird going on, I see the same hiding behaviour. btw that log is for blueman-manager ;-).

Looking into this.

infirit, 2017-02-01:
The offending commit that shows the strange icon behaviour is c0d7626b37f29d4a9cc32b074dbb2a20e4466df2. Now the challenge of understanding why 😄

infirit, 2017-02-03:
I am tired of fighting pygobject and have reverted this with 28db59370655ed0cb5918a2cac77c9c8f9341c99. Let me know if the problem persists.

closed 2017-02-03 by infirit

parkerlreed, 2017-02-03:
Turning back on the icon disappears for a second but does come back up. Seems to be fine here.

cschramm, 2017-02-03:
That glitch should be fixed in #599.