blueman-project/blueman #654: Logitech Mouse M555b

rowilug, 2017-02-01:
I want to use my logitech mouse M555b on my Asus R558U with Linux Mint 18 Sarah. I have installed blueman version 5.37. Whatever I do the mouse is not recognized by a scan........Need help

parkerlreed, 2017-02-01:
Run `bluetoothctl` in your terminal and try these commands

````
scan on #wait for mouse to be found
pair <mac adress> #you can tba complete the MAC address
trust <mac address>
connect <mac address>
````

If it also fails there then the issue doesn't lie with blueman. If it works there then something is up.

rowilug, 2017-02-02:
Hi thanks for your suggestion.

scan on does not find the mouse, trying a second time scan on returns 
the error below

Do you have any other suggestion ?

Best regards
- Roger -

parkerlreed, 2017-02-03:
Then this might be on the bluex end. Also what error did you get? Nothing came through from the email to website.

cschramm, 2017-02-03:
Did you try a completely different system? The mouse might even not be discoverable.

rowilug, 2017-02-03:
Hi,

Sorry I attached a screen-shot on the error and did not realize that 
it'll not come trough.

To summarize:
the first "scan on" returns without detection
a second "scan on" returns the error message "failed to start discovery: 
org.bluez.Error.InProgress"

Regards

###########################################################################################

rowilug, 2017-02-03:
The mouse was used before on an other computer and worked well, it is 
also discovered by my smartphone.

Regards

##############################################################################################################"

cschramm, 2017-02-03:
So you're able to discover the mouse with your phone while at the same time blueman and bluetoothctl are not able to discover it? Must be some issue in BlueZ, so that's out of our scope.

Maybe you see what's going on when you stop your bluetoothd and start it manuelly with `bluetoothd -d -n` before trying to discover the device, but maybe nothing will happen there as well.

rowilug, 2017-02-03:
Hi,

Thanks, that's what I get with bluetoothd -d -n

bluetoothd[12611]: Bluetooth daemon 5.37
bluetoothd[12611]: src/main.c:parse_config() parsing main.conf
bluetoothd[12611]: src/main.c:parse_config() Key file does not have key 
'DiscoverableTimeout' in group 'General'
bluetoothd[12611]: src/main.c:parse_config() Key file does not have key 
'PairableTimeout' in group 'General'
bluetoothd[12611]: src/main.c:parse_config() Key file does not have key 
'AutoConnectTimeout' in group 'General'
bluetoothd[12611]: src/main.c:parse_config() Key file does not have key 
'Name' in group 'General'
bluetoothd[12611]: src/main.c:parse_config() Key file does not have key 
'Class' in group 'General'
bluetoothd[12611]: src/main.c:parse_config() Key file does not have key 
'DeviceID' in group 'General'
bluetoothd[12611]: src/main.c:parse_config() Key file does not have key 
'ReverseServiceDiscovery' in group 'General'
D-Bus setup failed: Connection ":1.366" is not allowed to own the 
service "org.bluez" due to security policies in the configuration file
bluetoothd[12611]: Unable to get on D-Bus

Any new idea ?

##############################################################################################################

cschramm, 2017-02-03:
That didn't work. Not running as root? Also do not forget to stop the bluetooth service first.

You should not be able to start a discovery without bluetoothd running.

rowilug, 2017-02-03:
Bluetooth service is stopped
This is what I get by running as root

roger@roger-X556UQ ~ $ sudo bluetoothd -d -n
bluetoothd[14149]: Bluetooth daemon 5.37
bluetoothd[14149]: src/main.c:parse_config() parsing main.conf
bluetoothd[14149]: src/main.c:parse_config() Key file does not have key 
'DiscoverableTimeout' in group 'General'
bluetoothd[14149]: src/main.c:parse_config() Key file does not have key 
'PairableTimeout' in group 'General'
bluetoothd[14149]: src/main.c:parse_config() Key file does not have key 
'AutoConnectTimeout' in group 'General'
bluetoothd[14149]: src/main.c:parse_config() Key file does not have key 
'Name' in group 'General'
bluetoothd[14149]: src/main.c:parse_config() Key file does not have key 
'Class' in group 'General'
bluetoothd[14149]: src/main.c:parse_config() Key file does not have key 
'DeviceID' in group 'General'
bluetoothd[14149]: src/main.c:parse_config() Key file does not have key 
'ReverseServiceDiscovery' in group 'General'
D-Bus setup failed: Name already in use
bluetoothd[14149]: Unable to get on D-Bus

Regards

############################################################################################################################

cschramm, 2017-02-03:
You probably did not stop the service. I guess `sudo systemctl stop bluetooth` should do that on your system.

rowilug, 2017-02-03:
That's the result

roger@roger-X556UQ ~ $ sudo systemctl stop bleutooth
[sudo] password for roger:
Failed to stop bleutooth.service: Unit bleutooth.service not loaded.
roger@roger-X556UQ ~ $ sudo bluetoothd -d -n
bluetoothd[21811]: Bluetooth daemon 5.37
bluetoothd[21811]: src/main.c:parse_config() parsing main.conf
bluetoothd[21811]: src/main.c:parse_config() Key file does not have key 
'DiscoverableTimeout' in group 'General'
bluetoothd[21811]: src/main.c:parse_config() Key file does not have key 
'PairableTimeout' in group 'General'
bluetoothd[21811]: src/main.c:parse_config() Key file does not have key 
'AutoConnectTimeout' in group 'General'
bluetoothd[21811]: src/main.c:parse_config() Key file does not have key 
'Name' in group 'General'
bluetoothd[21811]: src/main.c:parse_config() Key file does not have key 
'Class' in group 'General'
bluetoothd[21811]: src/main.c:parse_config() Key file does not have key 
'DeviceID' in group 'General'
bluetoothd[21811]: src/main.c:parse_config() Key file does not have key 
'ReverseServiceDiscovery' in group 'General'
D-Bus setup failed: Name already in use
bluetoothd[21811]: Unable to get on D-Bus

#########################################################################################################################

rowilug, 2017-02-03:
Sorry I made a mistake, see the result below

roger@roger-X556UQ ~ $ sudo systemctl stop bluetooth
roger@roger-X556UQ ~ $ sudo bluetoothd -d -n
bluetoothd[22683]: Bluetooth daemon 5.37
bluetoothd[22683]: src/main.c:parse_config() parsing main.conf
bluetoothd[22683]: src/main.c:parse_config() Key file does not have key 
'DiscoverableTimeout' in group 'General'
bluetoothd[22683]: src/main.c:parse_config() Key file does not have key 
'PairableTimeout' in group 'General'
bluetoothd[22683]: src/main.c:parse_config() Key file does not have key 
'AutoConnectTimeout' in group 'General'
bluetoothd[22683]: src/main.c:parse_config() Key file does not have key 
'Name' in group 'General'
bluetoothd[22683]: src/main.c:parse_config() Key file does not have key 
'Class' in group 'General'
bluetoothd[22683]: src/main.c:parse_config() Key file does not have key 
'DeviceID' in group 'General'
bluetoothd[22683]: src/main.c:parse_config() Key file does not have key 
'ReverseServiceDiscovery' in group 'General'
D-Bus setup failed: Name already in use
bluetoothd[22683]: Unable to get on D-Bus
roger@roger-X556UQ ~ $

@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@

cschramm, 2017-02-04:
Looks like bluetoothd is still running. Not sure why. Look at you running processes to find out.

rowilug, 2017-02-04:
If I stop bluetooth using the gui or using the command "|sudo systemctl 
stop bluetooth" ,| these are the processes I can still find:

roger@roger-X556UQ ~ $ ps -aux | grep blue
roger     2172  0.0  0.1 150344  9948 ?        S    févr.01   0:00 
/usr/lib/bluetooth/obexd
roger     5504  0.0  0.6 557208 54852 ?        Sl   févr.01   0:03 
/usr/bin/python3 /usr/bin/blueman-applet
root     29220  0.0  0.0  32056  4672 ?        Ss   15:17   0:00 
/usr/lib/bluetooth/bluetoothd
roger    29272  0.0  0.0  14228   940 pts/0    S+   15:18   0:00 grep 
--color=auto blue

*****************************************************************************************************************************************************

cschramm, 2017-02-07:
So bluetoothd does *not* get stopped or automatically restarted in some way. You could try killing the process or temporarily disabling the service with `systemctl`. Otherwise check the logs why it's not stopped or restarted.

closed 2017-04-30 by cschramm