blueman-project/blueman #661: Bluetooth dies after a short period of time.

calismoonelf, 2017-02-13:
I asked this over in AskUbuntu about 6 weeks ago:

My BT freezes up while I'm using it. It actually comes back to life if I put the laptop to sleep and wake it up, but to be fair I have a connect script in /etc/pm/sleep.d from when 16.04 was not waking my network up. If I leave the "About" dialog box open in blueman-manager, it does not lock up. I have disabled the BT script in /usr/lib/pm-utils/power.d to see if that helped, it did not.

A workaround for a while was to open up blueman-manager, open the about dialog and then just move them both to another workspace where they are out of the way....this is no longer working, I have changed nothing other than doing regular updates.

Possible pertinent hardware info:

    Processor : 8x Intel(R) Core(TM) i7-4710MQ CPU @ 2.50GHz
    Memory : 12233MB
    Operating System : Ubuntu 16.10
    Network controller : Intel Corporation Wireless 3160 (rev 83)
    HP Bluetooth Laser Mobile Mouse

My connect wake-up script:

> #!/bin/sh
> 
> case "${1}" in
>         resume|thaw)
>         nmcli nm sleep false
>          systemctl restart networking
>          bluetoothctl connect 00:0F:F6:6D:46:38
>  ;;
> esac

I would be happy to generate any logs or reports to figure this out. Thank you!

infirit, 2017-02-13:
Can you describe that you mean with "die", obviously if the window is still there it did not die. My guess is it froze and you can not interact with it anymore.

Does the same happen to ``blueman-applet`` (the blue bluetooth icon in he tray/notification area)?

The logs to watch are the console output from blueman-manager and blueman-applet. Kill all of them and then start from a terminal.

calismoonelf, 2017-02-14:
When I say die, I mean that my bluetooth mouse becomes non-responsive. Following your instructions, I killed both of those processes. After killing them, my mouse was still functioning. So, this leads me to believe that the problem is not with one of those. But just in case, I still recorded the output from starting the up. 

-------

blueman-applet
blueman-applet version 2.0.4 starting
Stale PID, overwriting
_________
Load (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:60)
['NetUsage', 'DiscvManager', 'AppIndicator', 'GameControllerWakelock', 'NMDUNSupport', 'DBusService', 'NMPANSupport', 'TransferService', 'Menu', 'AuthAgent', 'StatusIcon', 'StandardItems', 'DhcpClient', 'PowerManager', 'Headset', 'ExitItem', 'KillSwitch', 'SerialManager', 'Networking', 'PPPSupport', 'RecentConns', 'ShowConnected'] 
/usr/lib/python3/dist-packages/blueman/plugins/applet/AppIndicator.py:8: PyGIWarning: AppIndicator3 was imported without specifying a version first. Use gi.require_version('AppIndicator3', '0.1') before import to ensure that the right version gets loaded.
  from gi.repository import AppIndicator3 as girAppIndicator
_________
get_interface_version (/usr/lib/python3/dist-packages/blueman/bluez/BlueZInterface.py:13)
Detected BlueZ 5 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.Menu.Menu'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.DBusService.DBusService'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.SerialManager.SerialManager'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
Not loading PPPSupport because it's conflict has higher priority 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.PowerManager.PowerManager'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.KillSwitch.KillSwitch'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'> 
_________
update_menuitems (/usr/lib/python3/dist-packages/blueman/plugins/applet/DiscvManager.py:123)
warning: Adapter is None 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.RecentConns.RecentConns'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
Not loading DhcpClient because it's conflict has higher priority 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.NMPANSupport.NMPANSupport'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.ExitItem.ExitItem'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.StandardItems.StandardItems'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.Headset.Headset'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.NMDUNSupport.NMDUNSupport'> 
_________
[... 125 lines cut ...]
Caching new device 10:4F:A8:98:E1:BC 
_________
row_update_event (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceList.py:264)
row update event Fake False 
_________
row_update_event (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceList.py:264)
row update event Trusted 1 
_________
row_update_event (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceList.py:264)
row update event Paired 1 
_________
Generate (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:173)
HP Bluetooth Laser Mobile Mouse 
_________
pa_context_event (/usr/lib/python3/dist-packages/blueman/main/PulseAudioUtils.py:341)
3 
_________
pa_context_event (/usr/lib/python3/dist-packages/blueman/main/PulseAudioUtils.py:341)
4 
_________
on_pa_ready (/usr/lib/python3/dist-packages/blueman/plugins/manager/PulseAudioProfile.py:29)
connected 
_________
<lambda> (/usr/lib/python3/dist-packages/blueman/main/PulseAudioUtils.py:353)
1

cschramm, 2017-02-15:
So your issue is that the mouse stops working. Does it get disconnected?

calismoonelf, 2017-02-21:
blueman and bluetoothctl both show it as connected but it becomes non-responsive. It is getting worse. I can close the laptop for just a few seconds and then log back in and it comes back.

cschramm, 2017-04-28:
You can check the connection status with `info 10:4F:A8:98:E1:BC` in `bluetoothctl` but I doubt that blueman just missed the disconnect. I guess it's just an issue with the mouse itself. Just to make sure I'd check the syslog for any messages like `mousedev` or `psmouse` to what goes on there (if anything at all) when it becomes unresponsive and when you close the laptop (i.e. suspend, right?).

cschramm, 2017-04-28:
Looks like #664 is pretty much the same for a Microsoft mouse. The referenced StackExchange thread might have a solution for you.

closed 2017-04-28 by cschramm