blueman-project/blueman #662: 'No Bluetooth Adaptor found' n 'No default controller available'

Umair-Anwar, 2017-02-17:
At GUI shows no adaptor available (Pi-3, service started)
in command line when power on or scan on then it says 'No default controller available'

thanks for concern,
umair

infirit, 2017-03-28:
I do not know enough about the rpi but the error means the adapter/bluetooth controller is not recognized by bluez. When bluez properly recognized blueman will (usually) happily manage it for you.

Maybe you are missing firmware? Check you kernel logs for hints.

closed 2017-04-30 by cschramm

FDiskas, 2018-01-18:
Same problem for me. I was using my Bluetooth for connecting headset, but because if low sound quality I tried the blueman project tools. ANd after playing around with a interface I got the same error message.
And I'm not able to connect my headphones any more. :poodle:
```ssh
$ sudo lsmod |grep blue
bluetooth             544768  12 btrtl,btintel,bnep,btbcm,btusb
ecdh_generic           24576  1 bluetooth
```
```bash
$ hcitool dev
Devices:
```
```bash
$ lsusb | grep Bluetooth
Bus 002 Device 005: ID 0cf3:3005 Atheros Communications, Inc. AR3011 Bluetooth
```
```
$ lspci -knn | grep Net -A2; lsusb
25:00.0 Network controller [0280]: Qualcomm Atheros AR9285 Wireless Network Adapter (PCI-Express) [168c:002b] (rev 01)
	DeviceName: WLAN
	Subsystem: Hewlett-Packard Company AR9285 Wireless Network Adapter (PCI-Express) [103c:1461]
	Kernel driver in use: ath9k
	Kernel modules: ath9k, wl
Bus 002 Device 005: ID 0cf3:3005 Atheros Communications, Inc. AR3011 Bluetooth
Bus 002 Device 004: ID 04f2:b230 Chicony Electronics Co., Ltd Integrated HP HD Webcam
Bus 002 Device 003: ID 046d:c52f Logitech, Inc. Unifying Receiver
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 046d:0a1a Logitech, Inc. 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 003: ID 138a:003c Validity Sensors, Inc. VFS471 Fingerprint Reader
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
:closed_lock_with_key::clock130::arrow_right_hook:

cschramm, 2018-01-25:
@FDiskas: Sounds like you're lacking firmware for your device or something fails to load. In both cases, you should find some information from ath3k in `dmsg` after connecting the device.

FDiskas, 2018-01-25:
After cold restart it works again

AriaMoradi, 2020-05-01:
this happens to me randomly when I hibernate the system

AriaMoradi, 2020-05-01:
open this again please