blueman-project/blueman #667: Active Icon Appears Twice

yodaldevoid, 2017-02-26:
I recently did a distro update for Ubuntu and after updating Blueman's tray icon was doubled. It is not a second icon next to the other, but rather the icon has two bluetooth symbols side by side. The attached picture should show what I mean. [image] All other icons appear as expected. I am running Ubuntu 16.10 with i3 as my window manager and i3status as my status bar.

infirit, 2017-03-28:
I'll setup a 16.10 vm and see if I can reproduce what iso did you use as base, unity, gnome, kde? in the mean time can you go into bluemans plugin dialog and disable the AppIndicator plugin and report if one of the two goes away.

yodaldevoid, 2017-03-28:
I believe there has been some misunderstanding. There is only one icon object, but the icon image has two Bluetooth symbols side by side. I have seen this issue on unity and i3.

infirit, 2017-03-28:
I understand. Blueman comes with two plugins for this icon. One that uses the old tray icon functionality the other is using the app indicator. Currently the former is mandatory the latter is not. It may be that for some reason this causes an issue and as it is a easy thing to try until I have vm setup :-).

Also of note is that blueman does not provide a black icon only blue ones. Can you try changing your icon theme to, lets say the default gtk one (adwaita) so that the blue one like [1] shows? Does it still look the same?

https://github.com/blueman-project/blueman/blob/master/data/icons/main_icon/hicolor_apps_128x128_blueman.png

yodaldevoid, 2017-03-28:
When I turn off the AppIndicator plugin nothing changes for my current i3 setup. It seems that Ubuntu was setting the icon them to ubuntu-mono-dark without me realizing, and when forced to Adwaita the default blue icon appears.

yodaldevoid, 2017-03-28:
It appears as if Ubuntu supplies a SVG for the bluetooth-active icon that is also used for the Blueman tray icon. It makes me wonder if SVGs are not being handled correctly somewhere in the pipeline.

infirit, 2017-03-28:
Just to be clear, with the blue icon it is show correctly?

The thing is we do not force any format or size. We just tell the implementation (tray/indicator) the icon name to use and leave it to handle the rest. Then the implementation looks up the icon in the currently active icon theme. The standard icon we set is ``blueman-tray`` and if the ShowConnected plugin is enabled __and__ we have active connection(s) we use ``blueman-active``.

I now have i3 running with i3bar for the tray. I have trouble having it show icons of the tray, can you post your ``bar { bla bla }`` config so I can copy it?

yodaldevoid, 2017-03-28:
Yes, the blue icon is correctly showing itself. Make sure that your primary monitor is actually set as primary under xrandr or else the icon tray may not show up. [docs](https://i3wm.org/docs/userguide.html#_tray_output)

```
bar {
        status_command i3status
        tray_output primary
        position top
}
```

yodaldevoid, 2017-03-28:
This is looking more and more like a problem on i3's side. I have just booted into a live CD of Ubuntu 16.10, installed Blueman, and don't see the problem in Unity (though I was fairly certain that I had seen it before.)

infirit, 2017-03-28:
Maybe, or ubuntu 16.10s icon for blueman. Unfortunately ubuntu 16.10 painfully to work with (slow mostly) in VirtualBox so getting things done is hard. I'll continue tomorrow to get the icons to show in i3bar.

infirit, 2017-03-29:
Ok, so the lack of icons is due to Gtk3 bugs and had to increase the font size. Now I have tray icons \o/.

But then I do not see the same problem you are.. The vm is based on a standard ubuntu desktop cd and then using the stable repo from https://i3wm.org/docs/repositories.html. 

When no devices are connected I see a single black icon and when connected the blueman-active (blue and green). So indeed this sounds like an issue with your install or i3bar.

yodaldevoid, 2017-03-31:
With some messing around with the font size for the bar's font I get the icon to display correctly. It indeed looks like it is a problem with i3bar, or rather cairo which I believe they use for rendering. For future reference a minimum font size of 12 was required for the icon to display correctly. I am going to close this as it is not Blueman's fault/issue.

closed 2017-03-31 by yodaldevoid

infirit, 2017-03-31:
Or it is Gtk not able to handle sizes smaller than 12. I have seen icon size problems and gtk status icon mentioned in several posts and bug reports on the interwebs. But in any case I do agree that from bluemans point of view there is not much we can do.