blueman-project/blueman #669: "Timed out waiting for reply from agent" when pairing with an iPhone

hobarrera, 2017-03-14:
I'm trying to pair my laptop with my iPhone, but I'm getting a `Timed out waiting for reply from agent` error. Here's what I'm doing:

* Make both devices discoverable (just in case really; one ought to do).
* Run `blueman-manager`
* Search for devices.
* Select the iPhone.
* Right click, select "Trust" (again, just in case, though I've tried without this step before)
* Right click, select "Pair"
* These both happen at the same time:
  * iPhone prompts "Bluetooth pairing request", with "Cancel" and "Pair".
  * Laptop shows a notification: "Pairing request for Hugo's iPhone, bla bla, MAC, bla CODE".
* Tap pair on the iPhone's popup.

Attempting to initiate from the iPhone (eg: clicking my laptop in its list) is basically the same popup (iPhone) and notification (laptop), with the same result.

Pairing codes match on both ends.

Here's what little logging I have:
```
$ sudo journalctl --follow -u bluetooth
Mar 14 18:56:37 athena.barrera.io systemd[1]: Starting Bluetooth service...
Mar 14 18:56:37 athena.barrera.io bluetoothd[1056]: Bluetooth daemon 5.44
Mar 14 18:56:37 athena.barrera.io bluetoothd[1056]: Starting SDP server
Mar 14 18:56:37 athena.barrera.io bluetoothd[1056]: Bluetooth management interface 1.14 initialized
Mar 14 18:56:37 athena.barrera.io systemd[1]: Started Bluetooth service.
Mar 14 18:56:44 athena.barrera.io bluetoothd[1056]: Endpoint registered: sender=:1.22 path=/MediaEndpoint/A2DPSource
Mar 14 18:56:44 athena.barrera.io bluetoothd[1056]: Endpoint registered: sender=:1.22 path=/MediaEndpoint/A2DPSink
Mar 14 19:01:52 athena.barrera.io bluetoothd[1056]: Timed out waiting for reply from agent
Mar 14 19:06:34 athena.barrera.io bluetoothd[1056]: Timed out waiting for reply from agent
```

Pair attempts are the last two lines (each attempts give an extra line).

Both devices pair up with my headphones fine, so bluetooth itself is working.

infirit, 2017-03-28:
Which version are you using and on which distro. The agent is part of ``blueman-applet`` and its log may yield some hints to what is going on. See https://github.com/blueman-project/blueman/wiki/Troubleshooting#debugging and provide the logs from applet start till it times out.

hobarrera, 2017-03-28:
blueman 2.0.4
bluez 5.11
ArchLinux (amd64)
i3wm

Debug output (I'm not sure how to trim out the bold/color markup, but if you copy this into a file, and `cat` it, you'll get the pretty output just like I saw it. 😄 ). I ran `blueman-manager` from a terminal.

```
_________
[1mLoad[0m (/usr/lib/python3.6/site-packages/blueman/main/PluginManager.py:60)
['Services', 'PulseAudioProfile'] 
_________
[1m__load_plugin[0m (/usr/lib/python3.6/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.manager.Services.Services'> 
_________
[1m__load_plugin[0m (/usr/lib/python3.6/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.manager.PulseAudioProfile.PulseAudioProfile'> 
_________
[1mpa_context_event[0m (/usr/lib/python3.6/site-packages/blueman/main/PulseAudioUtils.py:341)
1 
blueman-manager version 2.0.4 starting
Stale PID, overwriting
_________
[1mpa_context_event[0m (/usr/lib/python3.6/site-packages/blueman/main/PulseAudioUtils.py:341)
2 
_________
[1mon_bluez_name_owner_changed[0m (/usr/bin/blueman-manager:96)
org.bluez owner changed to  :1.5 
_________
[1mget_interface_version[0m (/usr/lib/python3.6/site-packages/blueman/bluez/BlueZInterface.py:13)
Detected BlueZ 5 
_________
[1mSetAdapter[0m (/usr/lib/python3.6/site-packages/blueman/gui/DeviceList.py:271)
 
_________
[1mon_adapter_changed[0m (/usr/lib/python3.6/site-packages/blueman/gui/manager/ManagerToolbar.py:83)
toolbar adapter /org/bluez/hci0 
_________
[1m__init__[0m (/usr/lib/python3.6/site-packages/blueman/main/Device.py:26)
caching initial properties 
_________
[1madd_device[0m (/usr/lib/python3.6/site-packages/blueman/gui/DeviceList.py:317)
adding new device 
_________
[1mdo_cache[0m (/usr/lib/python3.6/site-packages/blueman/gui/DeviceList.py:496)
Caching new device E0:B5:2D:DB:4F:A7 
_________
[1mrow_update_event[0m (/usr/lib/python3.6/site-packages/blueman/gui/manager/ManagerDeviceList.py:264)
row update event Fake False 
_________
[1mrow_update_event[0m (/usr/lib/python3.6/site-packages/blueman/gui/manager/ManagerDeviceList.py:264)
row update event Trusted 1 
_________
[1mrow_update_event[0m (/usr/lib/python3.6/site-packages/blueman/gui/manager/ManagerDeviceList.py:264)
row update event Paired 0 
_________
[1m__init__[0m (/usr/lib/python3.6/site-packages/blueman/main/Device.py:26)
caching initial properties 
_________
[1madd_device[0m (/usr/lib/python3.6/site-packages/blueman/gui/DeviceList.py:317)
adding new device 
_________
[1mdo_cache[0m (/usr/lib/python3.6/site-packages/blueman/gui/DeviceList.py:496)
Caching new device 00:07:04:CE:2C:3B 
_________
[1mrow_update_event[0m (/usr/lib/python3.6/site-packages/blueman/gui/manager/ManagerDeviceList.py:264)
row update event Fake False 
_________
[1mrow_update_event[0m (/usr/lib/python3.6/site-packages/blueman/gui/manager/ManagerDeviceList.py:264)
row update event Trusted 1 
[... 186 lines cut ...]
_________
[1mcheck[0m (/usr/lib/python3.6/site-packages/blueman/plugins/applet/PowerManager.py:119)
callbacks done 
_________
[1mset_adapter_state[0m (/usr/lib/python3.6/site-packages/blueman/plugins/applet/PowerManager.py:90)
True 
_________
[1m__init__[0m (/usr/lib/python3.6/site-packages/blueman/main/Device.py:26)
caching initial properties 
_________
[1m__del__[0m (/usr/lib/python3.6/site-packages/blueman/main/Device.py:72)
deleting device /org/bluez/hci0/dev_E0_B5_2D_DB_4F_A7 
_________
[1mDestroy[0m (/usr/lib/python3.6/site-packages/blueman/main/Device.py:94)
invalidating device /org/bluez/hci0/dev_E0_B5_2D_DB_4F_A7 
_________
[1mRequestConfirmation[0m (/usr/lib/python3.6/site-packages/blueman/main/applet/BluezAgent.py:205)
Agent.RequestConfirmation 
_________
[1mon_adapter_property_changed[0m (/usr/lib/python3.6/site-packages/blueman/plugins/applet/DiscvManager.py:96)
prop Discoverable 0 
_________
[1mon_adapter_property_changed[0m (/usr/lib/python3.6/site-packages/blueman/plugins/applet/DiscvManager.py:96)
Stop 
```

infirit, 2017-03-28:
So, my suspicion is that you are not seeing the notification requesting you to confirm the pair. Do you see a notification bubble? If not do you have a proper notification daemon running that supports actions?

If not install one or use blueman's fallback dialog window. To use the fallback change the key ``notification-daemon`` in gsettings schema ``org.blueman.general`` with dconf editor or use the following from the command line ``gsettings set org.blueman.general notification-daemon false``.

hobarrera, 2017-03-28:
Oh, now everything makes sense. :P

> So, my suspicion is that you are not seeing the notification requesting you to confirm the pair. Do you see a notification bubble? If not do you have a proper notification daemon running  [...]

Yup. I actually mentioned this in the initial report.

> [...] that supports actions?

Nope, it doesn't support actions, or other complex stuff. ~AFAIK, it's an optional capability.~ I just looked it up. [It's not](https://developer.gnome.org/notification-spec/).

TBH, now that I understand the full issue, I think this unveils a serious UX bug: looks like `blueman-manager` assumes that an external app will have an optional capability, and silently fails if it does not. This of course means a silent failure in lots of scenarios.

I've been having this problem for years, and I'm sure plenty of other have and never took the time to try and report/debug it, so maybe we should think a way around it (eg: when initiating a connection show a popup rather than an actionable notification).

infirit, 2017-03-29:
quick comment while im on the road. we ask the notification deamon for its capabilities and act accordingly. btw which one are you using?

so if the deamon replies properly we use it. when i am home ill post a dbus command that returns the data we check against.
-- 
Sent from my Android device with K-9 Mail. Please excuse my brevity.

infirit, 2017-03-29:
Back home 🎉 

The command to run from a terminal is ```gdbus call --session --dest org.freedesktop.Notifications --object-path /org/freedesktop/Notifications --method org.freedesktop.Notifications.GetCapabilities```. if in the output you see "actions" like mine below then the notifications deamon advertises it support them.

```
$ gdbus call --session --dest org.freedesktop.Notifications --object-path /org/freedesktop/Notifications --method org.freedesktop.Notifications.GetCapabilities       
(['actions', 'action-icons', 'body', 'body-hyperlinks', 'body-markup', 'icon-static', 'sound'],)
```

infirit, 2017-03-29:
btw, if you are using dunst it actually does support actions. But the feature is poorly documented, see https://github.com/blueman-project/blueman/issues/68#issuecomment-265852038

hobarrera, 2017-03-29:
Thanks for the multiple followups. I've had a crazy day and hadn't had a chance to reply:

* Yup, dunst. Caps are `(['actions', 'body', 'body-markup'],)`
* Extremely poorly documented: nor the README nor the man page mention anything related.

FWIW, it's currently being maintained (by new volunteers, AFAIK), but documentation is still a bit poor.

I'm still wondering if there's some sort of action that can be taken on `blueman`'s side or not. I mean, it was a silent failure, but, TBH, it's probably dunst's fault, so thats "okay". I don't think that asking via a notification when initiating the connection makes sense though; it's rare on an app that clicking a button asks for a confirmation via a notification.

infirit, 2017-03-29:
>it's currently being maintained (by new volunteers, AFAIK)

Excellent, it worked nicely when I looked into it for blueman. Took a bit of getting used to though 😸.

>I don't think that asking via a notification when initiating the connection makes sense though; it's rare on an app that clicking a button asks for a confirmation via a notification.

it has been this way on the major desktops since gnome2 and kde2 days. Not sure how gnome3 does it these days but plasma still uses a notification with actions. Not saying this is correct but at least for bluetooth pairing it has been the default. My guess is that it is like this as typically the agent is not part of the main ui as it is with blueman. What you interact with is blueman-manager but the agent is handled by blueman-applet which runs in the background with an icon in the tray.

hobarrera, 2017-03-29:
I, right, I get it; it's more of a complexity issue than a UX *choice*. 😝 

I had though that it was merely to reuse code when initiating and receiving a connecting (by using the same notification on both).

I guess that it's probably worth closing this then, since nobody's going to take the work around all this.

Thank!

infirit, 2017-03-31:
We have the fallback for folks who prefer a window. Maybe I'll see if I can put this this option in the ui ( plugin dialog probably) so it is easier to discover for people. But don't hold you breath I am quite busy right now 😃

closed 2017-03-31 by infirit

hobarrera, 2017-03-31:
I've started working on [improving the docs](https://github.com/dunst-project/dunst/pull/312) for `dunst`, which should avoid people bumping into this again (at least with dunst in particular).

Thanks!