blueman-project/blueman #670: Can't connect Sony Ericsson phone as an audio source

bodqhrohro, 2017-03-16:
If I try to use my SE W660i as an audio source, either from a manager or through a wizard, blueman for whatever reason requests the remote control too. So if I accept remote control on the phone and choose a profile, only remote control is working and the phone does not try to find and connect an audio device. If I decline, the connection just drops. When scanning for audio devices from the phone, it does not find any audio devices too. Maybe I need to tune something in BlueZ configuration to make PC act as the audio sink?

infirit, 2017-03-28:
Most of the audio issues are because of the missing pulseaudio bluetooth module, see the following entry containing two links to wikis that solve 90%+ of audio issues on linux.

infirit, 2017-03-28:
Euh, link https://github.com/blueman-project/blueman/wiki/Troubleshooting#a2dp-audio-device-paired-but-fails-to-connect

bodqhrohro, 2017-03-28:
@infirit those wikies seem to describe connecting external sinks to PC but I want PC to act as a sink instead.  However I have installed `pulseaudio-module-bluetooth` and restarted `pulseaudio` and `bluetooth` daemons but this did not change anything.

infirit, 2017-03-28:
The module is used in both directions. On android you have to make sure you have the option "media-audio" selected when connected with the linux host. Also try restarting to be sure, it shouldn't make a difference but you never know.

If I remember correctly the phone should also show up in pavucontrol does it for you?

closed 2017-06-07 by cschramm