blueman-project/blueman #673: Help with a review

Rudd-O, 2017-03-22:
Hello!  Can someone review https://github.com/blueman-project/blueman/pull/632 ?  infirit was assigned as a reviewer, and I've performed all the changes infirit requested, but now I am getting radio silence from the reviewer.

Thank you in advance!  :-)

cschramm, 2017-03-28:
It's on my radar. I'm sure @infirit will take another look at the current implementation. No hurry. We're moving slow. 😉

closed 2017-03-28 by cschramm

infirit, 2017-03-28:
I thought [this comment](https://github.com/blueman-project/blueman/pull/632#issuecomment-277024214) was clear how much more time I was going to invest in it (hint, this comment is already more than I wanted).

Rudd-O, 2017-03-28:
That's okay, someone else is taking a look at it, and as usual I will make the changes that are requested, like I did last time, even though you were gratuitously rude to me.