blueman-project/blueman #674: powerbeats3 "failed to add device"

1beb, 2017-03-29:
blueman: 2.0.4
BlueZ: 5.37-0ubuntu5
Distribution: Ubuntu 16.04
Desktop environment: Gnome

I can see the device and looks like I can connect to it, momentarily (I can see the animation at the left side of the device when I try to pair in the "bluetooth devices" window). But then it fails and says "failed to add device".

cschramm, 2017-05-02:
`Protocol not available` most probably means that the `pulseaudio-module-bluetooth` package is completely missing.

1beb, 2017-05-02:
Thank you, but no. It certainly is installed. This issue is still open, I'm not sure why it was closed. 

```
$ sudo apt-get install pulseaudio-module-bluetooth
[sudo] password for brandon: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
pulseaudio-module-bluetooth is already the newest version (1:8.0-0ubuntu3.2).
```

cschramm, 2017-05-02:
You can check if the modules are loaded with `pactl list modules`. The most relevant one would be `module-bluez5-discover`.

Searching for your error messages from bluetoothd also hints at an active killswitch. You should check `rfkill list`.

Your `blueman-applet` output seems incomplete; no pairing process visible. Can you try that again? Also, start `blueman-manager` in a separate terminal, not from the applet, to get seapate logs.

1beb, 2017-05-02:
`$ pactl list modules`

```
Module #26
	Name: module-bluez5-discover
	Argument: 
	Usage counter: n/a
	Properties:
		module.author = "João Paulo Rechi Vita"
		module.description = "Detect available BlueZ 5 Bluetooth audio devices and load BlueZ 5 Bluetooth audio drivers"
		module.version = "8.0"
```

`$ rfkill list`

```
0: hci0: Bluetooth
	Soft blocked: no
	Hard blocked: no
```

`$ blueman-applet` (startup only)

```
blueman-applet version 2.0.4 starting
_________
Load (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:60)
['NMDUNSupport', 'PPPSupport', 'SerialManager', 'StandardItems', 'DBusService', 'Menu', 'DhcpClient', 'StatusIcon', 'AuthAgent', 'DiscvManager', 'TransferService', 'KillSwitch', 'ShowConnected', 'RecentConns', 'NMPANSupport', 'NetUsage', 'GameControllerWakelock', 'ExitItem', 'Networking', 'PowerManager', 'Headset', 'AppIndicator'] 
_________
get_interface_version (/usr/lib/python3/dist-packages/blueman/bluez/BlueZInterface.py:13)
Detected BlueZ 5 
/usr/lib/python3/dist-packages/blueman/plugins/applet/AppIndicator.py:8: PyGIWarning: AppIndicator3 was imported without specifying a version first. Use gi.require_version('AppIndicator3', '0.1') before import to ensure that the right version gets loaded.
  from gi.repository import AppIndicator3 as girAppIndicator
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.Networking.Networking'> 
_________
load_nap_settings (/usr/lib/python3/dist-packages/blueman/plugins/applet/Networking.py:36)
Loading NAP settings 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.Headset.Headset'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.TransferService.TransferService'> 
_________
get_interface_version (/usr/lib/python3/dist-packages/blueman/bluez/obex/Base.py:20)
Detected BlueZ integrated obexd 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
Not loading DhcpClient because it's conflict has higher priority 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.Menu.Menu'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.PowerManager.PowerManager'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.KillSwitch.KillSwitch'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.SerialManager.SerialManager'> 
_________
[... 69 lines cut ...]
_________
on_registered (/usr/lib/python3/dist-packages/blueman/bluez/obex/AgentManager.py:18)
/org/blueman/obex_agent 

(blueman-applet:25001): Gdk-CRITICAL **: gdk_window_thaw_toplevel_updates: assertion 'window->update_and_descendants_freeze_count > 0' failed
_________
RequestPowerState (/usr/lib/python3/dist-packages/blueman/plugins/applet/PowerManager.py:131)
Requesting True 
_________
on_power_state_change_requested (/usr/lib/python3/dist-packages/blueman/plugins/applet/KillSwitch.py:118)
True 
_________
UpdatePowerState (/usr/lib/python3/dist-packages/blueman/plugins/applet/PowerManager.py:159)
off False 
foff False 
on True 
current state True 
new state True 
_________
check (/usr/lib/python3/dist-packages/blueman/plugins/applet/PowerManager.py:119)
callbacks done 
_________
set_adapter_state (/usr/lib/python3/dist-packages/blueman/plugins/applet/PowerManager.py:90)
True 
```

1beb, 2017-05-02:
Using the applet to connect: 

```
on_adapter_property_changed (/usr/lib/python3/dist-packages/blueman/plugins/applet/DiscvManager.py:96)
prop Discovering 1 
_________
on_adapter_property_changed (/usr/lib/python3/dist-packages/blueman/plugins/applet/DiscvManager.py:96)
prop Discovering 0 
org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
```

cschramm, 2017-05-02:
So BlueZ does not seem to answer the request. Do you get any output from bluetoothd right when you try to pair?

1beb, 2017-05-02:
This is all I can see otherwise: 

```
May 02 00:42:38 monster bluetoothd[1187]: Pair device timed out for hci0
May 02 00:45:53 monster bluetoothd[1187]: Pair device timed out for hci0
May 02 00:49:16 monster bluetoothd[1187]: Pair device timed out for hci0
```

cschramm, 2017-05-02:
By now I'm sure this isn't blueman related (i.e. you probably experience the same if you try to pair in `bluetoothctl` (after setting up an agent there)).

The message is a new one to me. No idea what might cause it other than the remote device just not answering. You can try to stop bluetoothd (e.g. using `systemctl`) and start it manually with the `-d` switch to get debugging output. You might need to kill `blueman-applet` and others to avoid getting the service restarted after stopping it. I'd also make sure the remote device is completely reset and does not possibly have any pairing information.

1beb, 2017-05-17:
Let's close this one. I've moved to a new computer and can no longer reproduce this exact problem.

closed 2017-05-17 by 1beb