blueman-project/blueman #675: Can't transfer files with Bluetooth

simon33-2, 2017-04-04:
Tried googling but couldn't find the answer very easily. So I'm sorry to bother you in advance if this proves a really easy fix.

The problem is that I can't turn on bluetooth file transfer on lubuntu 16.10 32bit.
blueman 2.04
bluez 5.41-0ubuntu3
lubuntu standard lxde desktop
[image]

If I go into "Local settings", the entire transfer panel is greyed out.
If I go into "Plugins", the "Transfer service" is unselected. If I select it, I get the attached error.

Hope that is enough for you to go on. I also can't browse/send from another computer.

simon33-2, 2017-04-04:
Oh hold on, if I run it as root from the command line, it works. It shouldn't be this hard though. The permissions should be set up properly on install.

infirit, 2017-04-04:
It is not a permission problem. Many distributions these days now using systemd do not enable obex file transfers globally. You will need to enable it for the user ``systemctl --user`` or globally ``systemctl --system``. I do not know the exact name of the service for your distro but on most it contains the word obexd.

closed 2017-04-04 by infirit

vranki, 2017-04-06:
How about:

 - Start obexd or whatever is required automatically.
 - If not possible, show a user-friendly message about the issue.
 - At least add scrollable and copy-pasteable error dialog.

Currently this still counts as a usability bug.

infirit, 2017-04-06:
>Currently this still counts as a usability bug.

I agree just not in blueman, I'll explain.

On non systemd distributions when the obex service is disabled it leaves the dbus service file for it alone. Systemd in its wisdom completely hides it for dbus and thus for blueman leaving us with no way to know if the dbus service is installed or just disabled. AFAIK we handle the non systemd case but I'll check later tonight.

infirit, 2017-04-07:
Checked current master (what will become version 2.1) and we do not fail like this any more. This is due to how we now handle dbus names appearing and vanishing with Gio's DBus implementation.

simon33-2, 2017-04-07:
Cool.

Not sure if this is a question for you but does this mean it will be fixed in lubuntu 17.04?

infirit, 2017-04-07:
depends on what you mean by fixed. If 17.04 comes with blueman 2.1 than it will not throw an error however you will still not be able to transfer files with no user feedback as to why. afaik gnome-bluetooth also does not give any user feedback when obexd is not available so with current blueman you at least get the python traceback indicating something is not right 😸