blueman-project/blueman #677: persistent org.bluez.Error.AuthenticationFailed on Debian Sid

markhkim, 2017-04-07:
am trying to connect Microsoft Surface Ergonomic Keyboard, without much success. I am running Debian Sid (+ GNOME 3) with the following bluetooth-related packages installed:
- `blueman` 2.1~alpha1-1
- `bluetooth` 5.43-2
- `bluez` 5.43-2
- `bluez-firmware` 1.2-3
- `firmware-misc-nonfree` 20161130-2
- `gnome-bluetooth` 3.20.1-1

With `scan on`, `bluetoothctl` discovers the keyboard. The following errors occur, however:

- Entering `pair` directly after discovery shows the following:

        Attempting to pair with [mac address]
        [CHG] Device [mac address] Connected: yes
        Failed to pair: org.bluez.Error.AuthenticationFailed
        [CHG] Device [mac address] Connected: no

- Entering `trust [mac address]` before pairing results in the same error message.
- Entering `pairable on` before pairing results in the same error message.
- Entering `connect [mac address]` shows the following:

        [CHG] Device [mac address] Connected: yes
        Failed to connect: org.bluez.Error.Failed
        [CHG] Device [mac address] Connected: no

- Entering `trust` or `pairable` before using `connect` results in the same error message.

Running `blueman-applet` results in the following Python error:

>     blueman-applet version 2.1.alpha1 starting
>     Traceback (most recent call last):
>       File "/usr/bin/blueman-applet", line 27, in <module>
>         BluemanApplet()
>       File "/usr/lib/python3/dist-packages/blueman/main/Applet.py", line 39, in __init__
>         self.DbusSvc = DbusService("org.blueman.Applet", "/")
>       File "/usr/lib/python3/dist-packages/blueman/main/DbusService.py", line 20, in __init__
>         self.bus = bus()
>       File "/usr/lib/python3/dist-packages/dbus/_dbus.py", line 211, in __new__
>         mainloop=mainloop)
>       File "/usr/lib/python3/dist-packages/dbus/_dbus.py", line 100, in __new__
>         bus = BusConnection.__new__(subclass, bus_type, mainloop=mainloop)
>       File "/usr/lib/python3/dist-packages/dbus/bus.py", line 122, in __new__
>         bus = cls._new_for_bus(address_or_type, mainloop=mainloop)
>     dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.

infirit, 2017-04-09:
blueman 2.1 alpha is not in sid. It is in experimental though so are you on sid or experimental? On sid with blueman 2.0.4 I can not replicate that traceback. I do see a timeout because bluetoothd is disabled but the dbus name is org.bluez is still shown on the bus.

```org.bluez.Error.AuthenticationFailed``` indicates that the pair was initiated but you have not confirmed the pin or the pin is incorrect. One normally should get a notification in gnome/console and on the phone, did you get one?

markhkim, 2017-04-11:
Oh right, I had experimental on. I downgraded to blueman 2.0.4 on Sid and got the following upon running `blueman-applet`:
https://gist.github.com/markhkim/6c92c08f6d14fe74f5cca4dafc442054

`org.bluez.Error.AuthenticationFailed` still shows up on `bluetoothctl` like I described above. I am trying to connect a bluetooth keyboard, so it cannot show a notification.

cschramm, 2017-04-11:
Try to register an agent in `bluetoothctl` (`agent on`) before pairing. Maybe it wants to tell you something and blueman fails to do so.

If that does not change anything `dmesg` might have some information and otherwise you can kill `bluetoothd` (you also need to stop things like blueman so that it does not automatically get started again) and start it manually with the `-d` option to get debugging output of the connection / pairing attempt.

markhkim, 2017-04-16:
`agent on` doesn't change the output on `bluetoothctl`: I still get `  Failed to pair: org.bluez.Error.AuthenticationFailed`.

Running `dmesg | grep blue` only returns the following, which I suppose is not very useful:
> [    3.704158] bluetooth hci0: firmware: direct-loading firmware brcm/BCM20702A1-0b05-180a.hcd

`killall bluetoothd` followed by `bluetoothd -d` results in `D-Bus setup failed: Name already in use`. `killall blueman` returns `blueman: no process found`.

cschramm, 2017-04-16:
If you're on systemd `systemctl stop bluetooth.service` should stop the daemon and keep it stopped even if something is still using the service.

There is no process named blueman. The main program is blueman-applet.

markhkim, 2017-04-23:
Bluetooth restarts right away after `systemctl stop bluetooth.service` (or `service bluetooth stop`, for that matter). Is there something else I need to kill other than `bluetoothd`?

markhkim, 2017-05-07:
@cschramm anything else I should try to produce debugging information?

cschramm, 2017-05-08:
Make sure that bluetoothd does not get restarted so that you can start it manually. Stop blueman and anything else that might access it beforehand. `systemctl disable bluetooth.service` could also do the trick if it does not work otherwise.

GameKyuubi, 2017-06-04:
I have the exact same problem pairing an Elecom M-BT13BL mouse.  No matter what options I use I get `Failed to pair: org.bluez.Error.AuthenticationFailed`

closed 2017-06-07 by cschramm

bieniekmateusz, 2017-11-29:
Did your bluetoothctl give you any pass code? If that is the case, you need to type that in on the keyboard and press enter. I did just that and could finally pair with the keyboard. Before that I kept receiving the same error you have. 

I assume it is not paired when you type in 'paired-devices'? 

Someone had a similar problem:
https://askubuntu.com/questions/701978/how-can-a-bluetooth-keyboard-that-requires-a-code-entry-be-paired-in-the-termina

yi-ji, 2018-10-18:
```
[bluetooth]# pair 00:12:6F:B6:11:FE
Attempting to pair with 00:12:6F:B6:11:FE
[CHG] Device 00:12:6F:B6:11:FE Connected: yes
Failed to pair: org.bluez.Error.AuthenticationFailed
[CHG] Device 00:12:6F:B6:11:FE Connected: no
```
Ubuntu 18, suddenly has this problem. (Both my keyboard and audio player worked fine yesterday)
also `dmesg` got these:

```
[  236.421844] dpc 0000:00:1c.0:pcie010: DPC containment event, status:0x1f00 source:0x0000
[  236.421879] pcieport 0000:00:1c.0: AER: Corrected error received: id=00e0
[  236.421899] pcieport 0000:00:1c.0: PCIe Bus Error: severity=Corrected, type=Data Link Layer, id=00e0(Transmitter ID)
[  236.421909] pcieport 0000:00:1c.0:   device [8086:a33d] error status/mask=00001000/00002000
[  236.421915] pcieport 0000:00:1c.0:    [12] Replay Timer Timeout  
[  239.256902] Bluetooth: hci0: last event is not cmd complete (0x0f)
[  255.179953] Bluetooth: hci0: last event is not cmd complete (0x0f)
```

infirit, 2018-10-18:
@ everyone who want to comment after this, don't. Open a new issue please.

@yi-ji, blueman requires a notification-deamon to display a notification with instructions. If you want to do this from ``bluetoothctl`` make sure to enable the agent with ``agent on``. If you still have problems see above.