blueman-project/blueman #678: systemd does not start bluetooth until an adapter is present

infirit, 2017-04-09:
This is on a fresh debian sid vm. We end up with an activatable name on the bus (it shows up in d-feet for example) but bluetoothd can not be started and the bus times out. I haven't tried master but wanted to get some discussion as systemd users will get hit with this in the next debian release (I think) if they have no bluetooth hardware (think usb dongle) and our applet crashes.

They insert the usb dongle and the applet does not appear in the tray. of course it will when they start blueman-manager but its not nice like that.

```
$ blueman-applet 
blueman-applet version 2.0.4 starting
Stale PID, overwriting
_________
Load (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:60)
['Headset', 'KillSwitch', 'DiscvManager', 'GameControllerWakelock', 'StatusIcon', 'ExitItem', 'NMDUNSupport', 'TransferService', 'Networking', 'NetUsage', 'AuthAgent', 'AppIndicator', 'NMPANSupport', 'RecentConns', 'DBusService', 'PowerManager', 'Menu', 'ShowConnected', 'DhcpClient', 'StandardItems', 'SerialManager', 'PPPSupport'] 
_________
Load (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:60)
Unable to load plugin module KillSwitch
Hardware kill switch not found 
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 175, in activate_name_owner
    return self.get_name_owner(bus_name)
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 361, in get_name_owner
    's', (bus_name,), **keywords)
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NameHasNoOwner: Could not get owner of name 'org.bluez': no such name

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/bin/blueman-applet", line 121, in <module>
    BluemanApplet()
  File "/usr/bin/blueman-applet", line 63, in __init__
    self.Plugins.Load()
  File "/usr/lib/python3/dist-packages/blueman/main/PluginManager.py", line 90, in Load
    __import__(self.module_path.__name__ + ".%s" % plugin, None, None, [])
  File "/usr/lib/python3/dist-packages/blueman/plugins/applet/AuthAgent.py", line 9, in <module>
    import blueman.main.applet.BluezAgent as BluezAgent
  File "/usr/lib/python3/dist-packages/blueman/main/applet/BluezAgent.py", line 23, in <module>
    from blueman.bluez.Agent import Agent, AgentMethod
  File "/usr/lib/python3/dist-packages/blueman/bluez/Agent.py", line 48, in <module>
    class Agent(dbus.service.Object):
  File "/usr/lib/python3/dist-packages/blueman/bluez/Agent.py", line 56, in Agent
    @AgentMethod
  File "/usr/lib/python3/dist-packages/blueman/bluez/Agent.py", line 38, in AgentMethod
    if BlueZInterface.get_interface_version()[0] < 5:
  File "/usr/lib/python3/dist-packages/blueman/bluez/BlueZInterface.py", line 16, in get_interface_version
    obj = dbus.SystemBus().get_object('org.bluez', '/')
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 241, in get_object
    follow_name_owner_changes=follow_name_owner_changes)
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 248, in __init__
    self._named_service = conn.activate_name_owner(bus_name)
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 180, in activate_name_owner
    self.start_service_by_name(bus_name)
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 278, in start_service_by_name
    'su', (bus_name, flags)))
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.TimedOut: Failed to activate service 'org.bluez': timed out
```

infirit, 2017-04-09:
With Gio we can use a flag so that it does not autostart and probably avoid this. I'll test this out but I did not see anything like that in dbus-glib so we may have to live with this untill 2.1 is released and in distros.

infirit, 2017-04-09:
For master see #678. Seems to work well and no timeout. Inserting the usb dongle makes blueman-applet show its tray icon. Also used it on my main machine and has no  negative impact so far.

cschramm, 2017-04-11:
Why aren't we able to activate the service? Is this due to `ConditionPathIsDirectory=/sys/class/bluetooth` in the service file? Makes sense, but I'd expect an error response rather than a timeout then. 🤔

infirit, 2017-04-11:
Not looked that closely but what I remember is that d-feet also timed out trying to activate the dbus service. and if i remember correctly it did not show a binary path and after usb dingle insertion it did.

infirit, 2017-04-12:
It seems like systemd just prevents autostart. However when I launch+kill bluetoothd __once__ manually as root it autostarts fine afterwards. Really weird, I just thought that I may want to check if starting bluetoothd creates /sys/class/bluetooth.

infirit, 2017-05-02:
Closing as #679 was merged

closed 2017-05-02 by infirit