blueman-project/blueman #680: Popup alter has wrong sizing in wayland

hobarrera, 2017-04-11:
When running blueman-manager under wayland, I get the usual startup popup, but instead of it's usual size it's *huge*.

This is probably just some window hint that's off:

[image]

infirit, 2017-08-11:
We don't set or enforce a size on the dialog. Can it be that your wm is doing this as it looks like you are using i3.

hobarrera, 2017-08-11:
The WM does this because the hints are off, and this is opened as a "normal window", rather than a popup (note: those might not be the exact terms in GTK-speak, but you get the idea, I hope).

infirit, 2017-08-11:
We use a GtkMessageDialog for this and we do not add any hints, it is a stock dialog provided by Gtk+. And for how we use it in this case its pretty much translates to a "normal window".

The type hint set by Gtk on a GtkMessageDialog is ``GDK_WINDOW_TYPE_HINT_DIALOG`` see https://developer.gnome.org/gdk3/stable/gdk3-Windows.html#GdkWindowTypeHint. I assume this then translates to _NET_WM_WINDOW_TYPE_DIALOG which accurately describes our dialog.

While this may be inconvenient on a tiling wm this is how Gtk handles dialogs.

closed 2017-11-21 by cschramm