blueman-project/blueman #681: PS3 ustable connection

brommeweb, 2017-04-13:
Hello

I having some problems with my connection to a PS3 Sixaxis controller. With the default versions of Blueman and BlueZ in Ubuntu XUbuntu 16.04.2 LTS I had an unstable connection that sometimes worked but most often blueman-manager show this error: Connection Failed:  Input/output error. By the help of google I found this issue  #422 this problem sounded similar to mine so I decided to install Blueman 2.1 Alpha, Unfortunatly this did not solve my issue, on the contrary it made it worse. I has only been able to connecte once, and since then I get same error.

I have executed blueman-report while trying to connect through the blueman-manager.

blueman: 2.1.alpha1
BlueZ: 5.37
Distribution: 
Desktop: XFCE

https://gist.github.com/c5263a12f15ebe734a37f1c17a28de7a#file-log-L181 Push button on PS3 controller 
https://gist.github.com/c5263a12f15ebe734a37f1c17a28de7a#file-log-L187 Connect though blueman-manager
https://gist.github.com/c5263a12f15ebe734a37f1c17a28de7a#file-log-L229 Push button again
https://gist.github.com/c5263a12f15ebe734a37f1c17a28de7a#file-log-L299 Click Always accept in Blutooth Authentication popup

My output from  bluetoothctl list
[NEW] Controller 00:26:83:10:E7:2A robot-top [default]
[NEW] Device 00:24:33:34:AD:8D PLAYSTATION(R)3 Controller
[CHG] Device 00:24:33:34:AD:8D Connected: yes
[CHG] Device 00:24:33:34:AD:8D Connected: no
[CHG] Device 00:24:33:34:AD:8D Connected: yes
[CHG] Device 00:24:33:34:AD:8D Connected: no

Can you please help with this issue, as my Linux skills have reached it limit now :)

cschramm, 2017-11-29:
Sorry for the veeery late reply, this one slipped through. There seems to be something wrong with your cairo setup as there're a lot of `TypeError: Couldn't find foreign struct converter for 'cairo.Context'`. Do you still have that issue?

brommeweb, 2017-11-29:
Hello Christoper

Yes I do believe since I put it aside a while ago as it did not work. I would very much like to take it up again and get it to work. I'm currently on my to USA for vacation and work so I get back to you when I return in 2 weeks time.

cschramm, 2017-12-05:
🤔 looks like my last comment got lost somehow. So, again: You're most probably just missing the `gi._gi_cairo` module which is part of PyGObject.

cschramm, 2018-01-07:
Closing for now. Reopen if you need any further assistance.

closed 2018-01-07 by cschramm