blueman-project/blueman #688: RFCOMM channel hard coded to 1

pideu-mh, 2017-05-03:
blueman: 2.0.4
BlueZ: 5.37
Distribution: Ubuntu 16.04.2 LTS
Desktop environment: Xubuntu

I am using a CSR8510 based Bluetooth dongle which is basically working fine.

Now I'd like to connect to a remote Bluetooth Classic device that uses RFCOMM on channel number 5.

Unfortunately this is not working correctly, a connection can never be established.

The root cause of the problem can be found in the _connect()_ method here:
[https://github.com/blueman-project/blueman/blob/master/blueman/services/meta/SerialService.py](url)
```
# TODO: Channel?
port_id = create_rfcomm_device(Adapter(self.device["Adapter"])['Address'], self.device["Address"], 1)
```

The comment already indicates that the author was well aware that hard coding the channel number to 1 is probably going to be a problem.

Anyway, in my local installation of _blueman_ I changed the 1 to 5 and restarted the _blueman-applet_. Afterwards I was able to connect to my remote Bluetooth Classic device that uses RFCOMM on channel number 5.

I have no idea of the inner workings of _blueman_, but my suggestion is to properly implement the _service discovery protocol_ (SDP) handling in order to find out the channel number the RFCOMM service on the remote device is using.

On the command line, that handling can be triggered using _sdptool_
```
$ sdptool records aa:bb:cc:dd:ee:ff
Service Name: Serial Port
Service RecHandle: 0x10000
Service Class ID List:
  "Serial Port" (0x1101)
Protocol Descriptor List:
  "L2CAP" (0x0100)
  "RFCOMM" (0x0003)
    Channel: 5
```

infirit, 2017-05-03:
>I have no idea of the inner workings of blueman

The low level bluetooth code is in https://github.com/blueman-project/blueman/blob/master/module/libblueman.c which needs some love to extract the channel. Then it is wrapped into a module for python with cython https://github.com/blueman-project/blueman/blob/master/module/_blueman.pyx.in

pideu-mh, 2017-05-09:
Thank you for your comment. 

Hm, in the files you mentioned the handling for the RFCOMM channel seems to be implemented correctly already.

In _create_rfcomm_device()_ a "channel" parameter can already be provided and in _def rfcomm_list():_ channel is appended as well.

It seems that the problem is in the higher layers instead.

In order to investigate this, I tried to upgrade my local installation to 2.1.alpha1-git by using the repository as mentioned in [https://github.com/blueman-project/blueman/wiki/Packaged-versions](url)

I can start _blueman-applet_ just fine, but when I do "Set up new device", then "Next", then select a device, then "Next", then "Pair Device", then "Next", then the applet crashes with the following message:

> Traceback (most recent call last):
>   File "/usr/bin/blueman-assistant", line 225, in on_prepare
>     device = self.Manager.find_device(self.Device['Address'], self.Adapter.get_object_path())
> AttributeError: 'NoneType' object has no attribute 'find_device'

I'm currently stuck here.

infirit, 2017-05-09:
>Hm, in the files you mentioned the handling for the RFCOMM channel seems to be implemented correctly already.

Yes, the missing part is figuring out the channel. And unfortunately rfcomm channel is not exposed in the bluez dbus api so we will some low level code, which is not really my area of expertise :smile:.

>I can start blueman-applet just fine, but when I do "Set up new device", then "Next", then select a device, then "Next", then "Pair Device", then "Next", then the applet crashes with the following message:

I'll have a look, thanks for reporting.

infirit, 2017-05-09:
@pideu-mh, Fixed the bug with blueman-assistant with commit 2b199f34192f810261546fc007535ea34b7fb2aa

pideu-mh, 2017-05-10:
Thanks for fixing the problem. I can now setup a new device correctly.

However, I have to disagree on this:

> Yes, the missing part is figuring out the channel. And unfortunately rfcomm channel is not exposed in the bluez dbus api so we will some low level code, which is not really my area of expertise 😄

From my perspective it looks like everything is already there.

In order to prove my point I forked the repo and added some proof-of-concept code. Let's continue the discussion in my pull request.

infirit, 2017-05-10:
>In order to prove my point I forked the repo and added some proof-of-concept code. Let's continue the discussion in my pull request.

Ah, yes good all is already there :+1:. Like I confessed not too familiar with serial and rfcomm, usually I wait for @cschramm :smiling_imp:.

cschramm, 2017-05-10:
> I have not yet understood how the service discovery procedure is triggered, who is then doing that service discovery procedure, how the information about the found devices then reaches blueman exactly and why the channel number does not seem to be part of that information.

Our primary interface is the BlueZ D-Bus API. Its `org.bluez.Device1` interface provides us with the `UUIDs` property which we use to identify services.

I guess libbluetooth's [sdp_lib.h](https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/lib/sdp_lib.h) is what should help us discover the channels.

infirit, 2017-05-30:
pybluez (gpl2) has some code to handle sdp. We could borrow from there or use pybluez for all the low level stuff?

cschramm, 2017-05-31:
We should probably change our SDP handling to something like [pybluez's do_search](https://github.com/karulis/pybluez/blob/master/bluez/btsdp.c#L75) does with sdp_lib.h instead of the BlueZ D-Bus API. I don't think adding pybluez as a dependency would be a good idea as we probably won't get rid of anything currently in libblueman.c (e.g. pybluez does *not* provide a way to create an RFCOMM device file) and the dependency would be a problem (e.g. both the Debian family and the openSUSE distributions currently seem to only include a package for Python 2).

infirit, 2017-09-23:
Looked on the internet and came up with #774. My c skills are not too good but it works for my bt spealer which exposes a serial service.

```python
>>> from _blueman import get_rfcomm_channel
# return 0 if it fails
>>> get_rfcomm_channel("00:9E:C8:61:C7:7F")
0
# turned on i get this
>>> get_rfcomm_channel("00:9E:C8:61:C7:7F")
24
```

infirit, 2017-09-24:
Updated #774 and it works nicely for me.

parkerlreed, 2017-09-25:
@infirit This is semi related and I figured I would ask first before posting a request if needed.

I have a Bluetooth device (serial port) that I currently use a Python script in a loop to: check if the device is broadcasting (on), and auto rfcomm bind if it is. 

Question is: would it be possible to maybe implement this as a plugin for Blueman? Idea is basically, "When device is in range, run arbitrary command (or run an action like binding to an rfcomm port)

infirit, 2017-10-04:
#774 merged, closing

closed 2017-10-04 by infirit