blueman-project/blueman #691: StatusIcon does not update with rfkill unblock 0

infirit, 2017-05-07:
When testing connman changes I used out statusicon to disable the adapter. Then I used ``rfkill unblock 0`` and the adapter started working again. However the statusicon remained as disabled.

Using the statusicon to enable and disable however works fine. My question is, are we supposed to handle this situation? I think we should though 😄.

cschramm, 2017-05-07:
Definitely yes. I'm not sure why we don't. My guess what happens is that something (I suppose udev) brings up the adapter automatically when you unblock it. What should happen then is:

1. `PropertyChanged` D-Bus signal for the `Powered` property of `org.bluez.Adapter1`
2. `BluemanApplet._on_adapter_property_changed`
3. `PowerManager.on_adapter_property_changed`
4. `PowerManager.UpdatePowerState`
5. `StatusIcon.IconShouldChange`
6. `StatusIcon.on_status_icon_resized`
7. `PowerManager.on_status_icon_query_icon` returns `None`
8. `StatusIcon.props.icon_name` gets set to an enabled icon

That chain looks good to me, so my guess is that point one never happens but that would be really strange... 🤔

infirit, 2017-05-08:
So, I enabled ``AutoEnable=true`` in bluez's main.conf and this is actually powering the adapter for me. I do see the property ``Powered`` change in bluetoothctl so we probably are at fault here. Marking as bug and I will investigate further.

cschramm, 2017-05-08:
Works for me in f34455f.

infirit, 2017-05-09:
that is annoying, it doesn't for me.

I'm building a fresh gentoo install and will check if i get the same in there.
-- 
Sent from my Android device with K-9 Mail. Please excuse my brevity.

cschramm, 2017-11-28:
Status?

infirit, 2017-11-28:
Not sure, will poke at it this weekend to confirm.

infirit, 2017-12-02:
Tracked it down to ``on_power_state_changed`` not being run on the applet plugins. It should be done in ``UpdatePowerState`` when ``self.current_state != new_state``.

Relevant bit from the debug log,

```
blueman-applet 00.08.02 INFO     KillSwitch:110 io_event  : killswitch changed 0
blueman-applet 00.08.02 INFO     KillSwitch:120 io_event  : State: True
blueman-applet 00.08.02 INFO     PowerManager:168 UpdatePowerState: off True | foff False | on True | current state False | new state False
blueman-applet 00.08.02 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0 {'Class': 1966340}
blueman-applet 00.08.02 DEBUG    DiscvManager:82 on_adapter_property_changed: prop Class 1966340
blueman-applet 00.08.02 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0 {'Powered': True}
blueman-applet 00.08.02 WARNING  PowerManager:193 on_adapter_property_changed: adapter powered on while in off state, turning bluetooth on
blueman-applet 00.08.02 INFO     PowerManager:118 RequestPowerState: Requesting True
blueman-applet 00.08.02 INFO     KillSwitch:136 on_power_state_change_requested: True
blueman-applet 00.08.02 DEBUG    KillSwitch:149 on_power_state_change_requested: Using mechanism to set state: True
blueman-applet 00.08.02 INFO     PowerManager:168 UpdatePowerState: off True | foff False | on True | current state False | new state False
blueman-applet 00.08.02 INFO     PowerManager:168 UpdatePowerState: off True | foff False | on True | current state False | new state False
blueman-applet 00.08.02 DEBUG    DiscvManager:82 on_adapter_property_changed: prop Powered True
blueman-applet 00.08.02 INFO     PowerManager:104 check     : callbacks done
blueman-applet 00.08.02 INFO     PowerManager:76 set_adapter_state: True
```

The problem is that ``new_state `` is still False. Its likely that ``off`` or ``foff`` is ``True`` but have not checked yet.

It may be because I only have usb bluetooth dongles that afak do not have a hardware killswitch.

infirit, 2017-12-03:
The main problem is that when the adapter powers on, ``on_adapter_property_changed`` on PowerManager does not immediately update the adapter state. 

Then when ``RequestPowerState`` is run it also does not update the adapter_state. In the end we run  ``UpdatePowerState`` with the old and wrong adapter_state.

See #801 for a potential fix. It is weird you are not seeing this :confused:.

cschramm, 2017-12-04:
There might be some race condition. What happens for me when `adapter_state` is `False` and I unblock the killswitch is:

1. Signal for the new interface -> `on_adapter_added` requests to set `Powered` to `False`
2. Signal `Powered` property changed to `True` -> `on_adapter_property_changed` triggers `RequestPowerState(True)`
3. Signal `Powered` property changed to `False` due to 1
4. `set_adapter_state(True)` due to the callback used in 2 -> requests to set `Powered` to `True`
5. Signal `Powered` property changed to `True` due to 4 -> `on_adapter_property_changed` triggers `RequestPowerState(True)`
6. `set_adapter_state(True)` due to the callback used in 5 -> requests to set `Powered` to `True` but that's not a change anymore.

2, 3, and 5 update the icon. 4 and 6 update the `adapter_state`. So 5 is the crucial step for me that updates the icon after 4 updated the `adapter_state`.

If 3 happens before 2 for you, 5 (and 6) would not happen as 4 does not change the `Powered` property anymore.

infirit, 2017-12-04:
Looking over your sequence I see that adapter_added is run, that does not happen for me. The adapter is never removed, just powered down and soft blocked. If so then this is the key difference, i only have a software block on my usb adapter.

```
# rfkill 
ID TYPE      DEVICE    SOFT      HARD
 0 bluetooth hci0   blocked unblocked
```

cschramm, 2017-12-04:
Interesting. Mine is only soft-blocked as well.

infirit, 2017-12-04:
Indeed, so it is just the adapter_added method you have and not me. I pushed a new approach to the PR which should take care of all adapters.

infirit, 2018-01-20:
Fixed by a6e34dd794b74d6ee14f64f655d8e96405d67389 :tada:

closed 2018-01-20 by infirit