blueman-project/blueman #696: Monochrome icons in top bar of GNOME Shell

PanderMusubi, 2017-05-12:
Please use monochrome icons when blueman is in the top bar of GNOME Shell. This is already the case when Bluetooth is switched on, but not when it is switched off. Perhaps use the one for when it is switched on with small x in the bottom right. See also attached screenshots.

[image]

PanderMusubi, 2017-05-12:
Use monochrome icons in the pull down menu and, if possible, in entire application when used in GNOME Shell. GNOME has enough stock icons so this does not require design work and the icons can be used from the GNOME Shell installation and don't need shipping. Fall back are the icons as they are now.

cschramm, 2017-05-12:
See #697. blueman is a generic tool not targeting anything like GNOME Shell specifically. It's up to the distributions and themes to change icons if they want to have different ones. Maybe that's already the case in the second image (no idea why it switches positions then though) but whoever did the work did not continue with the disabled icon.

closed 2017-05-12 by cschramm

PanderMusubi, 2017-05-12:
Then that is probably what is happening, but only one icon is replaced. what do you think is the best place to have GNOME follow up on this? Ubuntu?

cschramm, 2017-05-12:
Yes, you can file bugs against all those incomplete Ubuntu icon themes. I found the following in yakkety:

`numix-icon-theme`, `ubuntu-mate-icon-themes`, and `xubuntu-icon-theme` seem to have complete icon sets for blueman.

`elementary-icon-theme`, `faenza-icon-theme`, `humanity-icon-theme`, `mate-icon-theme-faenza`, `ubuntu-mono` (I guess that's what you're using), and `ubuntukylin-theme` only provide `blueman-tray`  but neither `blueman-active` nor `blueman-disabled`. That's probably from the days of blueman 1.

`lubuntu-icon-theme` lacks `blueman-disabled` as well. I'm not sure why it does have `blueman-active`... All versions either had both (blueman 2 and 1.99) or none (earlier versions).