blueman-project/blueman #697: Do not change places in top bar of GNOME Shell

PanderMusubi, 2017-05-12:
When the status of Bluetooth changes (on or off) the icon in the top bar of GNOME Shell changes places, see also attached screenshots. Please let it remain on the same position as this is less disturbing for the user.

[image]

This was reported for blueman version 2.0.4-1ubuntu2

cschramm, 2017-05-12:
The one below is not our logo so I suppose it's not blueman at all.

closed 2017-05-12 by cschramm

PanderMusubi, 2017-05-12:
Well, if you click on it, you get blueman. :-)

cschramm, 2017-05-12:
Looks like GNOME Shell does not maintain the position of applications then. I don't think there's anything we can do about that.

PanderMusubi, 2017-05-12:
Other application do remain in place. I think the change of icon might be the cause of it.

cschramm, 2017-05-12:
I'd file a bug against `gnome-shell` and await the reply.