blueman-project/blueman #698: Name full website url in about dialog

PanderMusubi, 2017-05-12:
Please name full website url in about dialog. See also attached screenshot. Some people want to see the url without having to create unwanted HTTP traffic.

[image]

cschramm, 2017-05-12:
You can right-click and copy it, no?

PanderMusubi, 2017-05-12:
Yes, I see that now. I would not expect that. But url look nicer IMHO.

infirit, 2017-05-19:
I care very little how it looks but we can do the below. Objections @cschramm?

[image]

cschramm, 2017-05-19:
Fine for me. 🙂

commit a14b3208 (2017-07-05): Set website label on the about dialog to the website url

closes #698
closed 2017-07-05 by infirit