blueman-project/blueman #701: error on start + crash after connected on BT audio adapter

scippio, 2017-05-18:
**on start:**
[image]
```
g-io-error-quark: GDBus.Error:org.freedesktop.DBus.Python.dbus.exceptions.DBusException: Not authorized (36)
```
in terminal:
```
 blueman-applet 
blueman-applet version 2.1.alpha1 starting
Stale PID, overwriting
blueman-applet 13.29.55 WARNING  PluginManager:152 __load_plugin: Not loading DhcpClient because its conflict has higher priority
blueman-applet 13.29.55 WARNING  PluginManager:152 __load_plugin: Not loading PPPSupport because its conflict has higher priority
Gtk-Message: GtkDialog mapped without a transient parent. This is discouraged.

(blueman-applet:26541): Gdk-CRITICAL **: gdk_window_thaw_toplevel_updates: assertion 'window->update_and_descendants_freeze_count > 0' failed

```

**On connecting:**
[image]
in terminal:
```
blueman-manager version 2.1.alpha1 starting
Stale PID, overwriting
/usr/lib/python3.6/site-packages/gi/overrides/GObject.py:553: Warning: gsignal.c:2641: instance '0x25b5100' has no handler with id '1555'
  return func(*args, **kwargs)
blueman-applet 13.30.34 WARNING  DiscvManager:119 update_menuitems: warning: Adapter is None
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xa0 in position 0: invalid start byte
blueman-applet 13.30.34 WARNING  DiscvManager:119 update_menuitems: warning: Adapter is None
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xe0 in position 0: invalid continuation byte
Traceback (most recent call last):
  File "/usr/lib/python3.6/site-packages/blueman/gui/DeviceList.py", line 64, in on_adapter_removed
    self.SetAdapter()
  File "/usr/lib/python3.6/site-packages/blueman/gui/DeviceList.py", line 237, in SetAdapter
    self.Adapter = self.manager.get_adapter(adapter)
  File "/usr/lib/python3.6/site-packages/blueman/bluez/Manager.py", line 86, in get_adapter
    raise ValueError("No adapter(s) found")
ValueError: No adapter(s) found
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xb0 in position 0: invalid start byte
UnicodeDecodeError: 'utf-8' codec can't decode byte 0x80 in position 0: invalid start byte
Traceback (most recent call last):
  File "/usr/lib/python3.6/site-packages/blueman/gui/manager/ManagerDeviceList.py", line 208, in on_finished
    super(ManagerDeviceList, self).device_remove_event(device, tree_iter)
  File "/usr/lib/python3.6/site-packages/blueman/gui/DeviceList.py", line 204, in device_remove_event
    sig = self.__signals.pop(device.get_object_path())
KeyError: '/org/bluez/hci0/dev_88_C6_26_A9_8C_3C'
blueman-manager 13.30.35 WARNING  DeviceList:142 update    : stopping monitor (row does not exist)
/usr/lib/python3.6/site-packages/blueman/gui/manager/ManagerProgressbar.py:164: Warning: Source ID 377 was not found when attempting to remove it
  GLib.source_remove(self.gsource)
```

Bluetooth service is on.
+ my main.conf (/etc/bluetooth/main.conf):
```
[General]

# Default adapter name
# Defaults to 'BlueZ X.YZ'
#Name = BlueZ

# Default device class. Only the major and minor device class bits are
# considered. Defaults to '0x000000'.
#Class = 0x000100

# How long to stay in discoverable mode before going back to non-discoverable
# The value is in seconds. Default is 180, i.e. 3 minutes.
# 0 = disable timer, i.e. stay discoverable forever
#DiscoverableTimeout = 0

# How long to stay in pairable mode before going back to non-discoverable
# The value is in seconds. Default is 0.
# 0 = disable timer, i.e. stay pairable forever
#PairableTimeout = 0

# Automatic connection for bonded devices driven by platform/user events.
# If a platform plugin uses this mechanism, automatic connections will be
# enabled during the interval defined below. Initially, this feature
# intends to be used to establish connections to ATT channels. Default is 60.
#AutoConnectTimeout = 60

# Use vendor id source (assigner), vendor, product and version information for
# DID profile support. The values are separated by ":" and assigner, VID, PID
# and version.
# Possible vendor id source values: bluetooth, usb (defaults to usb)
#DeviceID = bluetooth:1234:5678:abcd

# Do reverse service discovery for previously unknown devices that connect to
# us. This option is really only needed for qualification since the BITE tester
# doesn't like us doing reverse SDP for some test cases (though there could in
# theory be other useful purposes for this too). Defaults to 'true'.
#ReverseServiceDiscovery = true

# Enable name resolving after inquiry. Set it to 'false' if you don't need
# remote devices name and want shorter discovery cycle. Defaults to 'true'.
#NameResolving = true

# Enable runtime persistency of debug link keys. Default is false which
# makes debug link keys valid only for the duration of the connection
# that they were created for.
#DebugKeys = false

# Restricts all controllers to the specified transport. Default value
# is "dual", i.e. both BR/EDR and LE enabled (when supported by the HW).
# Possible values: "dual", "bredr", "le"
#ControllerMode = dual

# Enables Multi Profile Specification support. This allows to specify if
# system supports only Multiple Profiles Single Device (MPSD) configuration
# or both Multiple Profiles Single Device (MPSD) and Multiple Profiles Multiple
# Devices (MPMD) configurations.
# Possible values: "off", "single", "multiple"
#MultiProfile = off

# Permanently enables the Fast Connectable setting for adapters that
# support it. When enabled other devices can connect faster to us,
# however the tradeoff is increased power consumptions. This feature
# will fully work only on kernel version 4.1 and newer. Defaults to
# 'false'.
#FastConnectable = false

# Default privacy setting.
# Enables use of private address.
# Possible values: "off", "device", "network"
# "network" option not supported currently
# Defaults to "off"
# Privacy = off

Enable=Source,Sink,Media,Socket

[Policy]
#
# The ReconnectUUIDs defines the set of remote services that should try
# to be reconnected to in case of a link loss (link supervision
# timeout). The policy plugin should contain a sane set of values by
# default, but this list can be overridden here. By setting the list to
# empty the reconnection feature gets disabled.
#ReconnectUUIDs=00001112-0000-1000-8000-00805f9b34fb,0000111f-0000-1000-8000-00805f9b34fb,0000110a-0000-1000-8000-00805f9b34fb

# ReconnectAttempts define the number of attempts to reconnect after a link
# lost. Setting the value to 0 disables reconnecting feature.
#ReconnectAttempts=7

# ReconnectIntervals define the set of intervals in seconds to use in between
# attempts.
# If the number of attempts defined in ReconnectAttempts is bigger than the
# set of intervals the last interval is repeated until the last attempt.
#ReconnectIntervals=1,2,4,8,16,32,64

# AutoEnable defines option to enable all controllers when they are found.
# This includes adapters present on start as well as adapters that are plugged
# in later on. Defaults to 'false'.
AutoEnable=true
```

infirit, 2017-05-18:
Which distro are you using?

>g-io-error-quark: GDBus.Error:org.freedesktop.DBus.Python.dbus.exceptions.DBusException: Not authorized (36)

This means that you are not allowed to execute methods over dbus for ``org.blueman.Mechanism``. We use policykit to handle this for us, see https://github.com/blueman-project/blueman/wiki/Integrations#policykit

The second error may be fixed by 9a7d65efe4f2c327d8f8296cb4262894621721ab

scippio, 2017-05-18:
I'm using Archlinux. I added some policykit rules and the first error disappeared. Thank you.
But real problem is the second.

I'm using last git version... so I think this not help: 9a7d65e

infirit, 2017-08-11:
I think I know why (my fault, see 30b070746f4ea12febc81cc5eb4f9f15f8ad0354). Can you try current master and let us know if you still have this problem.

If you do can you run ``blueman-manager`` with ``--loglevel debug`` and post the output to a [gist](https://gist.github.com/) and provide the link here?

scippio, 2017-08-18:
Still crashing when connecting...
https://gist.github.com/scippio/d861140b937d4ebc143540276f18671d

infirit, 2017-08-18:
Your bluez daemon is actually crashing, which makes the dbus api we rely on disappear.
```
srp 18 09:46:56 crom systemd[1]: bluetooth.service: Main process exited, code=dumped, status=11/SEGV
srp 18 09:46:56 crom systemd[1]: bluetooth.service: Unit entered failed state.
srp 18 09:46:56 crom systemd[1]: bluetooth.service: Failed with result 'core-dump'.
```

Can you try to use ``bluetoothctl`` to connect and check if that works or not.

scippio, 2017-08-22:
```
[bluetooth]# devices
Device 88:C6:26:A9:8C:3C Logitech BT Adapter
Device 00:23:01:14:26:5C Energy Sport
[bluetooth]# connect 88:C6:26:A9:8C:3C
Attempting to connect to 88:C6:26:A9:8C:3C
[CHG] Device 88:C6:26:A9:8C:3C Connected: yes
[CHG] Device 88:C6:26:A9:8C:3C ServicesResolved: yes
[CHG] Device 88:C6:26:A9:8C:3C Paired: yes
Failed to connect: org.bluez.Error.InProgress
Agent unregistered
[DEL] Controller 88:9F:FA:F5:05:3E crom [default]
Waiting to connect to bluetoothd...
```

scippio, 2017-10-19:
Btw: My colleagues with Ubuntu and Mint in most cases join without problems.

scippio, 2017-11-07:
I don't known how... but it's working now :scream: ... I still have blueman-applet version 2.1.alpha1.

closed 2017-11-07 by cschramm

kewur, 2021-10-09:
well it's not working for me. And this is what I have in the logs 
`monitor_power_levels (/usr/lib/python3/dist-packages/blueman/gui/DeviceList.py:185)
Failed to get power levels
Get connection info failed
Authentication Failed
`

I also have the polkit rules on that wiki under a file called `/etc/polkit-1/rules.d/90-blueman.rules`