blueman-project/blueman #705: Proposal, restore Gtk.ImageMenuItem for blueman version 2.1

infirit, 2017-05-20:
Our menus look really ugly by not aligning the icon all the way to the left. While deprecated in Gtk3 has not been removed in version 3 but it has in version 4. As it stands now as long as we are on Gtk3 we  should be able to keep using Gtk.ImageMenuItem.

So, my proposal for blueman 2.1 is to restore Gtk.ImageMenuItem and have a good discussion how we want to deal with menus/Gtk4 for blueman 2.2.

cschramm, 2017-05-21:
Yeah, why not. I guess `blueman.main.indicators.GtkStatusIcon` is the right place for that once it's in and it's GTK+ 3 only anyway.

infirit, 2017-07-03:
@cschramm, do you want me to wait for your statusicon/indicator work to be done? Or can I go ahead and do it this week 😄.

cschramm, 2017-08-14:
🙈 Go ahead. 😆

infirit, 2017-08-14:
😄 #749

infirit, 2017-08-20:
#749 merged, will open a new issue for Gtk4

closed 2017-08-20 by infirit