blueman-project/blueman #707: Bluetooth was working, now no adapter...

captainyoung, 2017-05-31:
My blue tooth stopped working. Blueman does not find the adapter.

cschramm, 2017-05-31:
https://github.com/blueman-project/blueman/wiki/Troubleshooting

nhamlin23, 2017-06-29:
Hello, I am using a MultiBlue Dongle (Bluetooth HID/Flash Drive) the USB part of the drive works but when I open my Bluetooth MGR and select adapters it says there isn't any... I have spent hours up hours trying to resolve this issue... any help would be greatly appreciated. Thanks in advance!

cschramm, 2017-06-29:
@nhamlin23: What are the results of your troubleshooting?

nhamlin23, 2017-06-29:
Here is what it shows...
root@kali:~# /etc/init.d/bluetooth status
● bluetooth.service - Bluetooth service
   Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; vendor preset: disabled)
   Active: active (running) since Wed 2017-06-28 23:06:36 EDT; 2h 37min ago
     Docs: man:bluetoothd(8)
 Main PID: 9099 (bluetoothd)
   Status: "Running"
    Tasks: 1 (limit: 4915)
   CGroup: /system.slice/bluetooth.service
           └─9099 /usr/lib/bluetooth/bluetoothd

Jun 28 23:06:36 kali systemd[1]: Stopped Bluetooth service.
Jun 28 23:06:36 kali systemd[1]: Starting Bluetooth service...
Jun 28 23:06:36 kali systemd[1]: Started Bluetooth service.
Jun 28 23:06:37 kali bluetoothd[9099]: Bluetooth daemon 5.43
Jun 28 23:06:37 kali bluetoothd[9099]: Starting SDP server
Jun 28 23:06:37 kali bluetoothd[9099]: Bluetooth management interface 1.14 …ized
Hint: Some lines were ellipsized, use -l to show in full.

I have tried stopping and restarting the Bluetooth service

nhamlin23, 2017-06-29:
@cschramm I have tried finding out what chipset the device uses but I haven't had any luck with that either.

cschramm, 2017-06-29:
There is a "no adapter" section that also directs you to lsusb to find out the hardware ID: https://github.com/blueman-project/blueman/wiki/Troubleshooting#no-adapter-or-not-able-to-turn-on-bluetooth

nhamlin23, 2017-06-30:
@cschramm So this is what I found out about my device.. seems there isn't any information about the controller or chipset.. not sure what the next step should be.. any ideas? Thanks in advance!

[image]

cschramm, 2017-06-30:
Looks like vendor ID 1d54 is in fact completely unknown. You might just have to do some mode switching but I doubt that you will get it running, especially as there seems to be zero information about that device. This topic is clearly out of scope for blueman. You could [ask the BlueZ project](http://www.bluez.org/contact/) if you're in an adventurous mood.

(Closing the original topic as the author did not provide any information.)

closed 2017-06-30 by cschramm

nhamlin23, 2017-06-30:
@cschramm Thank you for your help/input sir!