blueman-project/blueman #715: Blueman-applet cannot restart bluetooth adapters

shiarus, 2017-06-30:
After turning bluetooth off (unexpectedly hangs, bug report already present), blueman cannot restart bluetooth adapters. Opening in terminal blueman-applet and trying again, but this produced no helpful errors, so i started blueman-adapters since this can only be opened when bluetooth adapters are started, this produced more helpful output:
pi@raspberrypi:~ $ blueman-adapters

** (blueman-adapters:3659): WARNING **: Error retrieving accessibility bus address: org.freedesktop.DBus.Error.ServiceUnknown: The name org.a11y.Bus was not provided by any .service files
blueman-adapters version 1.99.alpha1 starting
Gtk-Message: GtkDialog mapped without a transient parent. This is discouraged.
org.bluez owner changed to :1.7
Traceback (most recent call last):
  File "/usr/bin/blueman-adapters", line 232, in <module>
    BluemanAdapters(adapter_name)
  File "/usr/bin/blueman-adapters", line 42, in __init__
    check_bluetooth_status(_("Bluetooth needs to be turned on for the adapter manager to work"), lambda: exit())
  File "/usr/lib/python2.7/dist-packages/blueman/Functions.py", line 97, in check_bluetooth_status
    applet.SetBluetoothStatus(True, *args, **kwargs)
  File "/usr/lib/python2.7/dist-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib/python2.7/dist-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Python.IOError: Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/dbus/service.py", line 707, in _message_cb
    retval = candidate_method(self, *args, **keywords)
  File "<string>", line 2, in SetBluetoothStatus
  File "/usr/lib/python2.7/dist-packages/blueman/plugins/applet/PowerManager.py", line 192, in SetBluetoothStatus
    self.RequestPowerState(status)
  File "/usr/lib/python2.7/dist-packages/blueman/plugins/applet/PowerManager.py", line 133, in RequestPowerState
    self.UpdatePowerState()
  File "/usr/lib/python2.7/dist-packages/blueman/plugins/applet/PowerManager.py", line 180, in UpdatePowerState
    dprint("off", off, "\nfoff", foff, "\non", on, "\ncurrent state", self.current_state, "\nnew state", new_state)
  File "/usr/lib/python2.7/dist-packages/blueman/Functions.py", line 64, in dprint
    print s
IOError: [Errno 5] Input/output error

If i run ¨$ sudo /etc/init.d/bluetooth start¨ you will notice there exists no problem with the bluetooth service:
pi@raspberrypi:~ $ sudo /etc/init.d/bluetooth start
[ ok ] Starting bluetooth (via systemctl): bluetooth.service.

I have tried stopping and starting the service both before and after invoking blueman-applet or bluman-adapters. The result is always the same, bluetooth service reports ¨ok", blueman-applet plays ¨Turn Bluetooth On¨,and no bluetooth devices work. In fact, attempting to use a bluetooth device still causes the following error:
Failed to connect to device -
GDBus.Error:org.bluez.Error.NotReady:
Resource Not Ready. Try to connect again.

shiarus, 2017-06-30:
This is Bug #1699885 on LaunchPad, I opened it so it only has 6 heat.  However, this may be a clone of Bug # 411481 with 220 heat previously opened in 2009.

shiarus, 2017-06-30:
~ $ uname -a
Linux Asus-X540LA 4.4.0-79-generic #100-Ubuntu SMP Wed May 17 19:58:14 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

 ~ $ cat /proc/version
Linux version 4.4.0-79-generic (buildd@lcy01-30) (gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.4) ) #100-Ubuntu SMP Wed May 17 19:58:14 UTC 2017

 ~ $ cat /etc/issue
Linux Mint 18.1 Serena \n \l

cschramm, 2017-07-13:
Please first check your setup. Ubuntu 16.04 has blueman 2.0.4-1ubuntu2 but your output says 1.99.alpha1.

When your setup is fixed and if the problem still persists, please post the blueman-applet output of when you try to enable bluetooth, see [Troubleshooting](https://github.com/blueman-project/blueman/wiki/Troubleshooting#debugging).

closed 2017-08-24 by cschramm