blueman-project/blueman #718: ManagerDeviceList DeviceIcon issues (due to API changes and GDBus port)

infirit, 2017-07-01:
Now that I have been poking at the ManagerDeviceList for the icons I discovered some issues and bugs between 1.24 and 2.0. The current logic in the ManagerDeviceList does not match and work well with the BlueZ5 api. I went back to wheezy and had a look how it behaved.

* When a new device is discovered it has a "discovered" icon in the top right corner of the device icon. We never set this as we do not connect to the properties changed signal until after we added the device to the list. Subsequent searches do work and we add the discovered icon.
* In bluez4 we had to add devices and we would not allow setting the trusted property. There is no need to do this any more and the api allows any device to have its trusted property set. This breaks the discovered icon and we overwrite it with the trusted and if another search is issued we go back to the discovered version without trusted.

infirit, 2017-07-02:
My plan is to drop the discovered icon together with the icon opacity based of RSSI as there is no way to for us to differentiate between existing and new devices (if there is do tell as I can't find anything in the dbus api). 

I tried to think of a way to incorporate a transparent device icon somehow but I can't. One possible way is to have all unconnected devices (slightly?) transparent? Feel free to suggest something here 😄

closed 2017-07-05 by infirit

cschramm, 2017-08-14:
> there is no way to for us to differentiate between existing and new devices

That's my state of knowledge as well, yes.