blueman-project/blueman #722: Mesage: Network Manager failed to activate the connection

foggy67, 2017-07-05:
[image]

Hello
When pairing with my Samsung Galaxy Ace 4 to get the Internet connection from the PC, the message above appear.
Audio sharing to play an audio file from my smartphone works very well. Only the Internet link fails to set up.
Is this a bug?
Thank you.

parkerlreed, 2017-07-06:
NetworkManager also isn't working for me. It instantly drops connection even if I only do it from within NetworkManager.

The bnep0 interface devices gets created, never used, and I can manually run dhcpcd on it afterwards and it works fine.

````
Jul 06 11:35:20 inspiron NetworkManager[814]: <info>  [1499355320.7857] bluez: use BlueZ version 5
Jul 06 11:35:20 inspiron NetworkManager[814]: <info>  [1499355320.7870] manager: startup complete
Jul 06 11:35:20 inspiron NetworkManager[814]: <info>  [1499355320.7978] supplicant: wpa_supplicant running
Jul 06 11:35:20 inspiron NetworkManager[814]: <info>  [1499355320.8359] keyfile: add connection in-memory (31c96844-8303-4c7a-a1c2-dfc168b6bb37,"Nexus 5X Network")
Jul 06 11:35:20 inspiron NetworkManager[814]: <info>  [1499355320.8401] bluez: BT device Nexus 5X (A0:91:69:95:88:84) added (NAP)
Jul 06 11:35:20 inspiron NetworkManager[814]: <info>  [1499355320.8424] manager: (A0:91:69:95:88:84): new Bluetooth device (/org/freedesktop/NetworkManager/Devices/3)
Jul 06 11:35:20 inspiron NetworkManager[814]: <info>  [1499355320.8432] device (A0:91:69:95:88:84): state change: unmanaged -> unavailable (reason 'managed') [10 20 2]
Jul 06 11:35:20 inspiron NetworkManager[814]: <info>  [1499355320.8560] device (A0:91:69:95:88:84): state change: unavailable -> disconnected (reason 'none') [20 30 0]
Jul 06 11:37:37 inspiron NetworkManager[814]: <info>  [1499355457.7779] device (A0:91:69:95:88:84): Activation: starting connection 'Nexus 5X Network' (31c96844-8303-4c7a-a1c2-dfc168b6bb37)
Jul 06 11:37:37 inspiron NetworkManager[814]: <info>  [1499355457.7782] audit: op="connection-activate" uuid="31c96844-8303-4c7a-a1c2-dfc168b6bb37" name="Nexus 5X Network" pid=682 uid=1000 result="success"
Jul 06 11:37:37 inspiron NetworkManager[814]: <info>  [1499355457.7787] device (A0:91:69:95:88:84): state change: disconnected -> prepare (reason 'none') [30 40 0]
Jul 06 11:37:37 inspiron NetworkManager[814]: <info>  [1499355457.7792] manager: NetworkManager state is now CONNECTING
Jul 06 11:37:37 inspiron NetworkManager[814]: <info>  [1499355457.7803] device (A0:91:69:95:88:84): state change: prepare -> config (reason 'none') [40 50 0]
Jul 06 11:37:37 inspiron NetworkManager[814]: <warn>  [1499355457.7870] device (A0:91:69:95:88:84): Error connecting with bluez: GDBus.Error:org.bluez.Error.Failed: Operation already in progress
Jul 06 11:37:37 inspiron NetworkManager[814]: <info>  [1499355457.7871] device (A0:91:69:95:88:84): state change: config -> failed (reason 'bluetooth-failed') [50 120 44]
Jul 06 11:37:37 inspiron NetworkManager[814]: <info>  [1499355457.7875] manager: NetworkManager state is now DISCONNECTED
Jul 06 11:37:37 inspiron NetworkManager[814]: <warn>  [1499355457.7889] device (A0:91:69:95:88:84): Activation: failed for connection 'Nexus 5X Network'
Jul 06 11:37:37 inspiron NetworkManager[814]: <info>  [1499355457.8193] device (A0:91:69:95:88:84): state change: failed -> disconnected (reason 'none') [120 30 0]
Jul 06 11:37:45 inspiron NetworkManager[814]: <info>  [1499355465.1218] device (A0:91:69:95:88:84): Activation: starting connection 'Nexus 5X Network' (31c96844-8303-4c7a-a1c2-dfc168b6bb37)
Jul 06 11:37:45 inspiron NetworkManager[814]: <info>  [1499355465.1222] audit: op="connection-activate" uuid="31c96844-8303-4c7a-a1c2-dfc168b6bb37" name="Nexus 5X Network" pid=682 uid=1000 result="success"
Jul 06 11:37:45 inspiron NetworkManager[814]: <info>  [1499355465.1228] device (A0:91:69:95:88:84): state change: disconnected -> prepare (reason 'none') [30 40 0]
Jul 06 11:37:45 inspiron NetworkManager[814]: <info>  [1499355465.1233] manager: NetworkManager state is now CONNECTING
Jul 06 11:37:45 inspiron NetworkManager[814]: <info>  [1499355465.1244] device (A0:91:69:95:88:84): state change: prepare -> config (reason 'none') [40 50 0]
Jul 06 11:37:46 inspiron NetworkManager[814]: <info>  [1499355466.6565] device (A0:91:69:95:88:84): Activation: (bluetooth) Stage 2 of 5 (Device Configure) successful. Will connect via PAN.
Jul 06 11:37:46 inspiron NetworkManager[814]: <info>  [1499355466.6572] device (A0:91:69:95:88:84): state change: config -> ip-config (reason 'none') [50 70 0]
Jul 06 11:37:46 inspiron NetworkManager[814]: <info>  [1499355466.6631] dhcp4 (bnep0): activation: beginning transaction (timeout in 45 seconds)
Jul 06 11:37:46 inspiron NetworkManager[814]: ((src/platform/nm-platform.c:2650)): assertion '<dropped>' failed
Jul 06 11:37:46 inspiron NetworkManager[814]: g_array_sort_with_data: assertion 'array != NULL' failed
Jul 06 11:37:46 inspiron systemd[1]: NetworkManager.service: Main process exited, code=dumped, status=11/SEGV
Jul 06 11:37:46 inspiron systemd[1]: NetworkManager.service: Unit entered failed state.
Jul 06 11:37:46 inspiron systemd[1]: NetworkManager.service: Failed with result 'core-dump'.
Jul 06 11:37:46 inspiron systemd[1]: NetworkManager.service: Service hold-off time over, scheduling restart.
Jul 06 11:37:46 inspiron systemd[1]: Stopped Network Manager.
Jul 06 11:37:46 inspiron systemd[1]: Starting Network Manager...
````

NetworkManager seems to fail, then crash ad restart.

foggy67, 2017-07-06:
Thank you for your explanation, I am not alone dealing with that trouble :)
Can anybody solve these regressions or bugs? May be the package maintainer?

infirit, 2017-07-06:
Confirmed, something wonky is going on but have not checked if we are at fault.

infirit, 2017-07-22:
NetworkManager crashes with the following dbus call as well and with blueman not running. If folks are running into this file a bug report with your distro please.

```
$ gdbus call -y -d "org.freedesktop.NetworkManager" -o "/org/freedesktop/NetworkManager" -m "org.freedesktop.NetworkManager.ActivateConnection" "/org/freedesktop/NetworkManager/Settings/5" "/org/freedesktop/NetworkManager/Devices/6" "/"
```

infirit, 2017-07-23:
Should be fixed with NetworkManager 1.8.2, https://cgit.freedesktop.org/NetworkManager/NetworkManager/tree/NEWS?h=nm-1-8

foggy67, 2017-07-23:
Hello
Here is the result of the gbus command on my system:
damien@damien-Lenovo-G70-80:~$ gdbus call -y -d "org.freedesktop.NetworkManager" -o "/org/freedesktop/NetworkManager" -m "org.freedesktop.NetworkManager.ActivateConnection" "/org/freedesktop/NetworkManager/Settings/5" "/org/freedesktop/NetworkManager/Devices/6" "/"
Erreur : GDBus.Error:org.freedesktop.NetworkManager.UnknownConnection: Connection could not be found.
(According to introspection data, you need to pass 'ooo')

I have Network Manager 1.2.6 on my system.

infirit, 2017-07-24:
re-opening as the the @foggy67 actually has a different problem than me and @parkerlreed have.

@foggy67, that command line is from my machine and the object paths most likely are different for you.

Instead can you please follow https://github.com/blueman-project/blueman/wiki/Troubleshooting#debugging and provide the logs for __both__ ``blueman-applet`` and ``blueman-manager``.

foggy67, 2017-07-24:
Hi,
Here is the log for Blueman-applet, when I try to pair my 2 devices to get the internet connection from the PC:

damien@damien-Lenovo-G70-80:~$ blueman-applet --log-level debug
blueman-applet version 2.0.4 starting
Stale PID, overwriting
_________
Load (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:60)
['StandardItems', 'NMDUNSupport', 'RecentConns', 'SerialManager', 'GameControllerWakelock', 'Headset', 'AppIndicator', 'AuthAgent', 'ExitItem', 'ShowConnected', 'NetUsage', 'PPPSupport', 'NMPANSupport', 'KillSwitch', 'DiscvManager', 'Networking', 'TransferService', 'DhcpClient', 'PowerManager', 'Menu', 'DBusService', 'StatusIcon'] 
/usr/lib/python3/dist-packages/blueman/plugins/applet/AppIndicator.py:8: PyGIWarning: AppIndicator3 was imported without specifying a version first. Use gi.require_version('AppIndicator3', '0.1') before import to ensure that the right version gets loaded.
  from gi.repository import AppIndicator3 as girAppIndicator
_________
get_interface_version (/usr/lib/python3/dist-packages/blueman/bluez/BlueZInterface.py:13)
Detected BlueZ 5 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.Menu.Menu'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'> 
_________
update_menuitems (/usr/lib/python3/dist-packages/blueman/plugins/applet/DiscvManager.py:123)
warning: Adapter is None 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.DBusService.DBusService'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.NMDUNSupport.NMDUNSupport'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.NMPANSupport.NMPANSupport'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.TransferService.TransferService'> 
_________
get_interface_version (/usr/lib/python3/dist-packages/blueman/bluez/obex/Base.py:20)
Detected BlueZ integrated obexd 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.PowerManager.PowerManager'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.RecentConns.RecentConns'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.GameControllerWakelock.GameControllerWakelock'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.NetUsage.NetUsage'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.KillSwitch.KillSwitch'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.StandardItems.StandardItems'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.AppIndicator.AppIndicator'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
[... 248 lines cut ...]
_________
Generate (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:173)
SM-G357FZ 
_________
set_op (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:85)
op: regenerating instance <ManagerDeviceMenu.ManagerDeviceMenu object at 0x7f36fcc5daf8 (blueman+gui+manager+ManagerDeviceMenu+ManagerDeviceMenu at 0x2061340)> 
_________
set_op (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:85)
op: regenerating instance <ManagerDeviceMenu.ManagerDeviceMenu object at 0x7f36f60fc9d8 (blueman+gui+manager+ManagerDeviceMenu+ManagerDeviceMenu at 0x2061a90)> 
_________
unset_op (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:99)
op: regenerating instance <ManagerDeviceMenu.ManagerDeviceMenu object at 0x7f36fcc5daf8 (blueman+gui+manager+ManagerDeviceMenu+ManagerDeviceMenu at 0x2061340)> 
_________
Generate (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:173)
SM-G357FZ 
_________
unset_op (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:99)
op: regenerating instance <ManagerDeviceMenu.ManagerDeviceMenu object at 0x7f36f60fc9d8 (blueman+gui+manager+ManagerDeviceMenu+ManagerDeviceMenu at 0x2061a90)> 
_________
fail (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:124)
fail org.freedesktop.DBus.Python.dbus.exceptions.DBusException: Network Manager Failed to activate the connection 
/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerProgressbar.py:165: Warning: Source ID 821 was not found when attempting to remove it
  GObject.source_remove(self.gsource)

May those logs help you....

closed 2017-07-24 by foggy67