blueman-project/blueman #723: Fedora 26 beta, blueman error at bootup

karl69942, 2017-07-05:
g-io-error-quark: GDBus.Error:org.freedesktop.DBus.Python.GLib.Error: Traceback (most recent call last):
  File "/usr/lib64/python3.6/site-packages/dbus/service.py", line 707, in _message_cb
    retval = candidate_method(self, *args, **keywords)
  File "/usr/lib/python3.6/site-packages/blueman/main/DbusService.py", line 38, in wrapper
    return method(*args[1:], **kwargs)
  File "/usr/lib/python3.6/site-packages/blueman/plugins/mechanism/Network.py", line 56, in ReloadNetwork
    self.confirm_authorization(caller, "org.blueman.network.setup")
  File "/usr/libexec/blueman-mechanism", line 166, in confirm_authorization
    action_id, {}, 1, "")
  File "/usr/lib64/python3.6/site-packages/gi/overrides/Gio.py", line 172, in __call__
    None)
GLib.GError: g-dbus-error-quark: GDBus.Error:org.freedesktop.DBus.Error.AccessDenied: An SELinux policy prevents this sender from sending this message to this recipient, 0 matched rules; type="method_call", sender=":1.36" (uid=0 pid=1492 comm="python3 /usr/libexec/blueman-mechanism " label="system_u:system_r:blueman_t:s0") interface="org.freedesktop.PolicyKit1.Authority" member="CheckAuthorization" error name="(unset)" requested_reply="0" destination=":1.16" (uid=997 pid=810 comm="/usr/lib/polkit-1/polkitd --no-debug " label="system_u:system_r:policykit_t:s0") (9)
 (36)

kuboosoft, 2017-07-07:
Same problem in F26... I will modify the rpm, cross fingers!

infirit, 2017-07-07:
The problem is not with blueman. Fedora uses SELinux to limit root operations and blueman-mechanism is not allowed to access Policykit over dbus. You will want to file a bug on fedora to update the selinux policy.

closed 2017-07-07 by infirit