blueman-project/blueman #724: Bluetooth device neither recognized nor recognizing other bluetooth devices

sivaraam, 2017-07-06:
My laptop came with a Broadcom bluetooth adapter. When I try to enable
it gets enabled but doesn't show any nearby devices. 

I once tried the fix specified in [this blog](https://www.gnebehay.com/blog/lenovo-flexpad-bluetooth-debian). It worked for some time and after
some upgrade it stopped working again. I have re-installed the
operating system since then (not for this reason).

I am currently unable to use bluetooth to transfer files (or) connect devices.
The only thing I could do is to watch it "trying" to detect devices.
What could be the issue?

Information:

blueman: 2.0.4
BlueZ: 5.43-2
Laptop Vendor : DELL
Laptop Model : 3542
Distribution: Debian GNU/Linux buster/sid 64-bit
Kernel:  4.9.0-3-amd64
Display Manager : GNOME
Display server : X

sivaraam, 2017-07-06:
I have tried to ask for help for this issue in places like the debian-users mailing list, bluez mailing list and I didn't get any reply from either of them. So, I'm filing this issue in the hope that I would get help here. 

In case I'm in the wrong place, let me know what the right place is.

kuboosoft, 2017-07-07:
Hey! check your firmware...

sivaraam, 2017-07-13:
@kuboosoft Thanks for the info. It solved the issue almost completely. It just doesn't work out of the box as I have to do a `sudo modprobe -r btusb && sudo modprobe btusb` everytime to bring it live. It's not a big issue, though.

closed 2017-07-13 by cschramm