blueman-project/blueman #726: Use Appearance for caption and blueman device icon

infirit, 2017-07-07:
We can use the gatt Appearance property exposed to retrieve the type of device. It maps directly to the numbers listed at [1].

I am not sure in which module we should put this list, DeviceClass would be an option as this is closest to how we will use it.

[1] https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.characteristic.gap.appearance.xml

infirit, 2017-07-07:
Very hacky and rough patch https://github.com/blueman-project/blueman/compare/master...infirit:device_appearance?expand=1

My mouse now shows in the caption as "Mouse".

cschramm, 2017-07-13:
I dind't realize that those devices do not have the `Class` property. Your patch makes absolute sense now that I did. 😆

infirit, 2017-07-13:
Cool, I'll think about this a bit and open a PR when ready.

infirit, 2017-08-03:
#740 

If we get unexpected Appearance numbers we should check the spec list and update ours.

Should we unify all of this so we have a consistent caption?

infirit, 2017-09-11:
Merged #740

closed 2017-09-11 by infirit