blueman-project/blueman #730: can't change bluetooth adapter

xychen9459, 2017-07-22:
I have two bluetooth adapters connected to computer. When I use blueman 1.23, I can change between two adapters using blueman-manager. But after I upgrade blueman to 1.99, I can't change between adapters. How to solve the problem?

I install blueman using ubuntu apt-get with repository ppa:blueman/ppa.

infirit, 2017-07-22:
You do not mention which ubuntu version [1] you use but I guess it is 14.04.

1.99 was an alpha release which still had many problems. The latest released version is 2.0.4 which you should use instead, or stick with 1.24. Also the ppa is not maintained us and the latest update is 2 years ago, please consider it unmaintained for now.

@cschramm, not sure if you have access or know Liviu Andronic who uploaded the alpha, maybe we can close the ppa or at least get the 2.0.4 in there?

[1] next time please make sure you follow the link that appears on the top when creating a new issue.
https://github.com/blueman-project/blueman/blob/master/CONTRIBUTING.md

cschramm, 2017-07-22:
@infirit: What ppa?

infirit, 2017-07-22:
It is part of the old launchpad account, https://launchpad.net/~blueman/+archive/ubuntu/ppa

cschramm, 2017-08-24:
Gone 💥

closed 2017-08-26 by infirit