blueman-project/blueman #731: reporter-unfriendly stacktrace dialog in blueman-applet

krichter722, 2017-07-22:
I just encountered the following dialog in `blueman-applet`:

[image]

The text isn't copyable and there's no immediate information where to report the bug. The stacktrace reaches to the bottom of the page and it's unclear whether it's displayed completely or exceeds the screen (not visible on the screenshot because the screenshot application compensates it, but will be the case for many non-oversize displays).

Please note that the issue revealed by the dialog is not subject of the this issue report and that it's not a support request, but a request to fix the dialog so that not only the actual issue, but all can be reported in an efficient way.

experienced with 2.0.4-1ubuntu2 on Xubuntu 17.04

infirit, 2017-07-24:
The actual error is systemd being silly and is fixed in master.

We can easilly add a link to the issue tracker here but are you sure the text is not selectable? I ask because I remember it being selectable and I don't have a xenial vm setup at the moment.

infirit, 2017-07-24:
Ok, I see we have many error dialogs and this is not one with the nice expander and selectable label. I'll add a common error dialog and use it across blueman. See #733

infirit, 2017-08-20:
#733 merged, closing.

closed 2017-08-20 by infirit