blueman-project/blueman #734: TypeError: tuple indices must be integers, not dbus.String

AllenTuring, 2017-07-28:
I installed blueman from the Debian repositories via `sudo apt-get install blueman` along with bluez and bluetooth packages.

In the Local Services dialog,attempting to create a NAP returns the dialog "Failed to apply network settings" with the content "TypeError: tuple indices must be integers, not dbus.String"

Is there a fix available for this?

infirit, 2017-07-28:
Yes it is fixed in version 2.0.4 which is in jessie backports.

closed 2017-07-28 by infirit

AllenTuring, 2017-07-28:
Thanks, checked out the version on Jessie backports.

However, now the apply button is greyed out, not selectable. Is there documetation for creating a NAP with blueman?

infirit, 2017-07-28:
The Apply button is only active if something changed from before.

Is the NAP checkbox ticked? If so you should have a the selected dhcp daemon running and you are good to go.

AllenTuring, 2017-07-29:
I can tick and untick the NAP checkbox, and change all of the settings, but the Apply button remains greyed out. Even with the NAP ticked none of my other devices are detecting a Bluetooth PAN to join.

AllenTuring, 2017-07-30:
Okay, I know this is probably no longer in this project scope, but with dnsmasq I installed the default package and using [the debian dnsmasq guide](https://wiki.debian.org/HowTo/dnsmasq) as a reference, in dnsmasq.conf I changed:

* `#interface=` -> `interface=pan1`
* `#dhcp-range=192.168.0.50,192.168.0.150,12h` -> `dhcp-range=pan1192.168.0.50,192.168.0.150,12h`

`dnsmasq --test` returns that my changes are still syntactically valid, and `systemctl status ` for dnsmasq and bluetooth shows both services are active.

However Blueman's NAP creation throws [an error](http://pastebin.com/8rJcmpV9). Apologies if this is a bad dnsmasq setup on my part, I'm still fairly new at this. What would be a proper way to set up dnsmasq to serve the NAP?

infirit, 2017-07-30:
We don't use the config file and pass arguments directly on the command. If curious see [here](https://github.com/blueman-project/blueman/blob/2-0-stable/blueman/main/NetConf.py#L36)

The error means we tried starting dnsmasq but failed. You should also have a dnsmasq error message in your logs, check /var/log (or whatever systemd uses).

AllenTuring, 2017-07-30:
I reset the config file, although still experiencing the error. [These](https://pastebin.com/qD2bhLP8) are the most recent contents of /var/log/syslog. btpan-local is the hostname.

infirit, 2017-07-30:
>And I suppose, looking at the logs, the root cause of that is DNSMasq's failed to create listening socket for 169.254.112.36: Address already in use line?

Yes it looks like something else is already listening on that address (the previous isc dhcpd maybe?). Make sure all other dhcp daemons are stopped or if they need to be running not use the same address. To see what listens on that ip use ```netstat -ap | grep "169.254.112.36"```

AllenTuring, 2017-07-31:
I had uninstalled isc-dhcpd-server, so I don't think it's that. `netstat -ap | grep "169.254.112.36"` returns nothing, I tried looking at `netstat -ap` alone and [this is my active internet connections](https://pastebin.com/PGyA35R4) section. What I think this says:

* [Lines 3, 12, 15, 26](https://pastebin.com/KTTnRVBY) are the dnsmasq server at process id 685, from debian package dnsmasq.
* [Lines 4-11, 13](https://pastebin.com/kf733Z80) are the XRDP session (and supporting tightvncserver) for remotely accessing the"btuser" user on the "btpan-local" system, from a system LAPTOP, as well as the parallel SSH session open on port 677.
* [Lines 14, 22-24](https://pastebin.com/uSnRNERd) are process 437, Avahi daemon, which is running the network zeroconf/bonjour discovery service, and from package libnss-mdns.
* [Lines 17-21, 28-30](https://pastebin.com/6jj4ZTsE) are connections by PID 693, Network Time Protocol Daemon
* [Lines 16, 25, 31](https://pastebin.com/rCTWinZM) is DHCP daemon, process 455

There's also a huge section of Active UNIX domain sockets afterwards.

cschramm, 2017-08-24:
I guess `btpan-local` is the `169.254.112.36` (`netstat -n` gives you numbers). As you can see `dnsmasq` is bound to the NTP port. Maybe that's what causing the message in the log.