blueman-project/blueman #739: Bridge firewalling is disabled and need to load module br_netfilter

infirit, 2017-08-03:
>Aug  3 09:43:11 infirit kernel: [   29.763908] bridge: filtering via arp/ip/ip6tables is no longer available by default. Update your scripts to load br_netfilter if you need this.

My kernel version:
```
$ uname -a
>Linux infirit 4.12.2-gentoo #1 SMP Mon Jul 17 22:44:50 CEST 2017 x86_64 Intel(R) Core(TM) i7-3770K CPU @ 3.50GHz GenuineIntel GNU/Linux
```

This may be a custom kernel problem (I am on gentoo) but I had to load ``br_netfilter`` manually for NAP to work. But to be sure if folks are on kernel version on or close to 4.12 to check if they see the same message and need to load ``br_netfilter``.

cschramm, 2017-08-24:
I'm seeing that same message and it seems like loading that module has always been necessary since 3.18 (http://ebtables.netfilter.org/documentation/bridge-nf.html). Are you saying the NAT does not work?

infirit, 2017-08-25:
>Are you saying the NAT does not work?

Yup, without ``br_netfilter`` no data can be transmitted through the bridge. Just tested again and unloaded the module, nothing. Load the module and reload network and everything works perfectly.

cschramm, 2017-08-25:
Looks like we need to call `modprobe` together with the `iptables` calls then. 😕

infirit, 2017-08-26:
Just tried this on debian 9 and it seems it works without that module loaded. Very confused atm and it might be something on my end. Will reboot the vm and make sure I did not mess something up.

rworkman, 2017-11-23:
Maybe wrap the modprobe in a try:  -- it will return success whether the module was loaded by that or was already loaded; if it fails, then tell the user not to expect things to work :-)

infirit, 2019-04-20:
I don't have the time to investigate this right now so closing as most folks do not appear to run into this but me.

closed 2019-04-20 by infirit

raviktiwari007, 2020-07-29:
Not sure if something has changed recently at Ubuntu side... but I got an update for core_ububtu as part of apt-get upgrade and since then I am getting this error. Looks like there are many more people getting impacted with this:

My exact error is:
[   20.810428] bridge: filtering via arp/ip/ip6tables is no longer available by default. Update your scripts to load br_netfilter if you need this

My kernel details (uname -a):
Linux my.domain.com 4.15.0-112-generic #113-Ubuntu SMP Thu Jul 9 23:41:39 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

Just so you know this test server is installed on Windows 2016 Server -> VMware 15.5 Pro -> Ubuntu 18.04 LTS

I was wondering if it is worth opening this thread so that someone can have a detailed look into it.

Many Thanks,
Rav

raviktiwari007, 2020-07-29:
Not sure if something has changed recently at Ubuntu side... but I got an update for core_ububtu as part of apt-get upgrade and since then I am getting this error. Looks like there are many more people getting impacted with this:

My exact error is:
[ 20.810428] bridge: filtering via arp/ip/ip6tables is no longer available by default. Update your scripts to load br_netfilter if you need this

I am on "systemd 237" and my kernel details (uname -a) are:
Linux my.domain.com 4.15.0-112-generic #113-Ubuntu SMP Thu Jul 9 23:41:39 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

Just so you know this test server is installed on Windows 2016 Server -> VMware 15.5 Pro -> Ubuntu 18.04 LTS Server Edition (No GUI)

I was wondering if it is worth opening this thread so that someone can have a detailed look into it.

Many Thanks,
Rav