blueman-project/blueman #750: Systray icon misaligned in Arch Linux

Hacksaurus, 2017-08-16:
I was just wondering if this is an issue pertaining to my icon theme or this is a bug with bluman... But it seems to be some sort of misalignment in the bluetooth icon...

Examples:
[image]
[image]

From these 2 perspectives it seems like that blueman is causing the trouble...

Thanks for any help!

cschramm, 2017-08-16:
You mean it is not centered but shifted to the left a little bit?

That is not our icon. It comes from some icon theme.

I've listed icon themes that contain icons for blueman and are included in Ubuntu yakkety in https://github.com/blueman-project/blueman/issues/696#issuecomment-301047681.

Hacksaurus, 2017-08-16:
Oh, I'm sorry I have the icon pack "Numix-Circle"

cschramm, 2017-08-16:
My guess is that you're seeing https://github.com/numixproject/numix-icon-theme/blob/master/Numix/22/panel/bluetooth-active.svg, so it's not the icon itself that's off.

Are you using the AppIndicator plugin (check if a left click opens the menu (app indicator) or the manager window (otherwise))?

infirit, 2017-08-16:
It look like the numix icon is 18x22 which could explain it not aligning properly. Our default icons are 22x22.

in my quick test it shows the same behaviour in the classic tray and appindicator.

Hacksaurus, 2017-08-16:
@cschramm Whenever I click the menu, it opens the manager...

cschramm, 2017-08-16:
Looks like an issue in the Numix icon theme. All icons I checked in https://github.com/numixproject/numix-icon-theme/tree/master/Numix/22/panel are rectangular except the bluetooth ones (to which the blueman ones link to). I'd create an issue there, @Hacksaurus.

closed 2017-08-16 by cschramm

Hacksaurus, 2017-08-17:
Ok got it! Thanks for the help!