blueman-project/blueman #753: ManagerDeviceList call _set_device_cell_data

infirit, 2017-08-20:
While running with debug logging I noticed lots of calls to get_icon_info. I traced it to _set_device_cell_data function but this is strange.

Normally this should only be called when there are actual changes, like moving the window or moving the mouse cursor over it. This probably also means also that other parts of the TreeView get updated every second.

For comparison see https://gist.github.com/infirit/ceceb54fc872bdf312cae89b2f2a9915

CPU usage AFAICS is negligible but it is filling up the logs so I'll drop the log call until I figure out why.

infirit, 2017-08-20:
Found the issue quicker than I expected 😨. The primary reason for this is the unnessesary calls to set: https://github.com/blueman-project/blueman/blob/6c57718412f2ad7667e445388ca3598222935f51/blueman/gui/manager/ManagerDeviceList.py#L389

The fix is to only update on actual change but I am struggling with with the ``if True``: https://github.com/blueman-project/blueman/blob/6c57718412f2ad7667e445388ca3598222935f51/blueman/gui/manager/ManagerDeviceList.py#L332
I see no point to it and the else part will never be called, unless I am missing something obvious..

cschramm, 2017-08-21:
It has ever been like this (since d76ebf3e1). We can just drop the `if` clause and the `else` branch.

closed 2017-08-21 by infirit