blueman-project/blueman #762: Can not pair certain devices in GUI

sup2069, 2017-08-26:
Using the GUI to pair an Iphone 7 Plus with auto pair, and setup GUI, ends in with an "unsuccessful pair". One must use bluetoothctl's pair command in terminal to receive the pair code prompt, allowing for a then successful pair. 

GUI lacks this functionality, or is not working as intended.

blueman: 2.0.4
BlueZ: 5.45
Distribution: Debian 9 (stretch)
Desktop environment: Mate

Report tool debug upload
[log.txt](https://github.com/blueman-project/blueman/files/1254006/log.txt)

url
https://gist.github.com/anonymous/4262d4491984981ba30692cf43ba5e2a

cschramm, 2017-08-27:
Hi, I just added https://github.com/blueman-project/blueman/wiki/Troubleshooting#pairing-fails-on-the-remote-devices-side-wihout-a-pairing-dialog-showing for that case. You've already done the `bluetoothctl` step. Please try the python snippet and see what happens.

Please confirm that you have the package `mate-notification-daemon` and your `libnotify4` is 0.7.7.

sup2069, 2017-08-27:
Thanks for the prompt reply! I already have `mate-notification-daemon` and `libnotify4` (0.7.7-2) installed. Ran the python snippet and got the test notification to popup. 
[image]

cschramm, 2017-08-28:
Strange. This is the interesting part from your log:

```
[1mRequestConfirmation[0m (/usr/lib/python3/dist-packages/blueman/main/applet/BluezAgent.py:205)
Agent.RequestConfirmation 
```

It shows that blueman does get the pairing request from BlueZ. The code at that point is:

```
        dprint("Agent.RequestConfirmation")
        alias = self.get_device_alias(device)
        notify_message = _("Pairing request for:") + "\n%s" % alias
        if passkey:
            notify_message += "\n" + _("Confirm value for authentication:") + " <b>%s</b>" % passkey
        actions = [["confirm", _("Confirm")], ["deny", _("Deny")]]

        self.n = Notification("Bluetooth", notify_message, 0, actions, on_confirm_action,
                              pixbuf=get_icon("blueman", 48), status_icon=self.status_icon)
```

So it should lead directly to the notification that you just tested successfully.

Maybe the `(blueman-applet:6417): Gdk-CRITICAL **: gdk_window_get_origin: assertion 'GDK_IS_WINDOW (window)' failed` points to some issue (@infitit: Any idea?).

There are only two points where your test differs: It does not use an icon or provide position hints. To  test the icon you can do:

```
from blueman.gui.Notification import Notification
from blueman.Functions import get_icon
Notification("Title", "Message", 0, [["action1", _("Action 1")], ["action2", _("Action 2")]], pixbuf=get_icon("blueman", 48))
```

To test if the position hints are the problem, you can edit /usr/lib/python3/dist-packages/blueman/main/applet/BluezAgent.py. Remove the `status_icon` argument from line 222 so that it is just `pixbuf=get_icon("blueman", 48))`. Then restart `blueman-applet` and try pairing again.

infirit, 2017-08-28:
>Maybe the ```(blueman-applet:6417): Gdk-CRITICAL **: gdk_window_get_origin: assertion 'GDK_IS_WINDOW (window)' failed``` points to some issue (@infitit: Any idea?).

its harmless and you can ignore that. Its AppIndicator not having a place to put its indicators and being annoying about it.

sup2069, 2017-08-28:
> To test if the position hints are the problem, you can edit /usr/lib/python3/dist-packages/blueman/main/applet/BluezAgent.py. Remove the status_icon argument from line 222 so that it is just pixbuf=get_icon("blueman", 48)). Then restart blueman-applet and try pairing again.

Success! 

[image]

infirit, 2017-08-28:
Disabling AppIndicator plugin should also workaround this. For some reason get_geometry is way off when in this situation and we display the notification off screen.

cschramm, 2017-08-28:
@infirit: What exactly is "this situation"? Notifications work fine for me with AppIndicator on Xfce with xfce4-notifyd (might very well be that it just ignores the position hints).

infirit, 2017-08-28:
So when it is enabled it gives me x and y like this,
```blueman-applet 19.51.41 WARNING  StatusIcon:46 geometry  : Got garbage x 330320972 or y 32766 value```. Considering my screen is 1920x1080 x and y are not even close.

However, just looked into this a bit more. AppIndicator actually provides a fallback StatusIcon so we probably should just move things over. If you don't mind, for 2.0 I'll prepare a PR that checks x and y against the screen size and always return something sane. In my quick tests using zero seems to work ok.

infirit, 2017-08-28:
Workaround to try, https://github.com/infirit/blueman/commit/f7aa1fe5f3cbe3d2c9bb949df3bc7ee3c0910d1a

sup2069, 2017-08-29:
> Workaround to try, infirit/blueman@f7aa1fe

I reversed cschramm's edit to `/usr/lib/python3/dist-packages/blueman/main/applet/BluezAgent.py` and made the edits to `blueman/gui/Notification.py`.

Success! The only noticeable change is that the notification is now on the left instead of the right. I will report back once I reboot. 

[image]

sup2069, 2017-08-29:
Rebooted the computer, GUI pair successful again.

PR #764 (2017-09-05): Notification: Make sure x and y hint are in screen geometry
>We should probably use AppIndicator for status icon but I rather not
>do big changes in 2.0. So workaround it by returning zero when we get
>weird x and y values from get_geometry.

This fixes #762. I decided for a slightly different solution than what was posted in #762.
closed 2017-08-29 by sup2069