blueman-project/blueman #763: Adapter not found

jlovett, 2017-08-28:
My Bluetooth worked fine, until one day when it didn't.  I kept hoping it was just an update that had killed it by chance and that it would be fixed in the weeks to come, but that isn't what's happened.  My co worker and I have the same hardware and are both running Fedora 26.  His hasn't had any issues. I'm not sure if the adapter died, or if it is in fact a software fault.

cschramm, 2017-08-29:
https://github.com/blueman-project/blueman/wiki/Troubleshooting#no-adapter-or-not-able-to-turn-on-bluetooth

ncrohn, 2017-08-30:
I am having the same issue. When I run `bluetoothctl list` nothing returns. My bluetooth headset was working fine yesterday. Rebooted this morning and now it's not finding the adapter any more. The only thing I've done is system updates.

I'm running Ubuntu 16.04 on kernel 4.12.8.

cschramm, 2017-08-30:
@ncrohn:

(There's a conflict in your data. While there is an Ubuntu package with Linux 4.12.8 it's not part of any released version including 16.04.)

When the controller is missing, I'd check the killswitches (`rfkill list`) and `lspci`. If everything seems right, check the Kernel messages to see what the driver tells you.

closed 2017-09-11 by cschramm