blueman-project/blueman #765: Protocol not available

rwst, 2017-09-04:
blueman: blueman-manager version 2.1.alpha1
BlueZ:
Distribution: OpenSuSE Leap
Desktop environment: Xfce

* [X] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

> ℹ Please use some service like [Pastebin](http://pastebin.com/) or [GitHub Gist](https://gist.github.com/) to post your logs and keep the thread clean and readable. Make sure to describe the exact steps you took when producing it.

Since a system upgrade I cannot connect to audio with the app. It says immediately `Conection failed. Protocol not available`
Debug output:
```
blueman-manager version 2.1.alpha1 starting
Stale PID, overwriting
_________
notify (/usr/lib/python2.7/site-packages/blueman/plugins/applet/RecentConns.py:213)
<Device.Device object at 0x7f047b5725a0 (blueman+bluez+Device+Device at 0xeed470)> 0000110b-0000-1000-8000-00805f9b34fb 
_________
initialize (/usr/lib/python2.7/site-packages/blueman/plugins/applet/RecentConns.py:129)
rebuilding menu 
/usr/lib/python2.7/site-packages/blueman/gui/manager/ManagerProgressbar.py:163: Warning: Source ID 706 was not found when attempting to remove it
  GLib.source_remove(self.gsource)
```
My bluetoothctl list
```
[NEW] Controller 5C:F3:70:78:11:43 ark [default]
[NEW] Device F8:77:B8:25:73:24 [Samsung] Soundbar J-Series
[bluetooth]# quit
[DEL] Controller 5C:F3:70:78:11:43 ark [default]
```
Previously I got a pop-up after few seconds that asked me for permission to connect. This is no longer the case, connection fails at once.

cschramm, 2017-09-04:
I bet you need [pulseaudio-module-bluetooth](https://software.opensuse.org/package/pulseaudio-module-bluetooth).

rwst, 2017-09-05:
No cigar. I'll let you know when I find out.

cschramm, 2017-09-05:
🤔 That message definitely means that module is missing (just search for it in past issues reported here).

Check `pactl list | grep module-bluetooth-discover` and `sudo pactl load-module module-bluetooth-discover`.

rwst, 2017-09-06:
OK, my pulseaudio config was reset to non-functional---thanks SuSE.

closed 2017-09-06 by rwst