blueman-project/blueman #771: Bluetooth speaker does not connect - Philips BT6000

TxusLado, 2017-09-15:
blueman:
BlueZ:
Distribution:
Desktop environment:

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

> ℹ Please use some service like [Pastebin](http://pastebin.com/) or [GitHub Gist](https://gist.github.com/) to post your logs and keep the thread clean and readable. Make sure to describe the exact steps you took when producing it.

TxusLado, 2017-09-15:
I have installeg lubuntu in an old packard bell computer
Lubuntu runs well but bluetooh does not connect with the bluetooth speaker
My bluetooth dongle is an USB 2.0
Files can be shared by bluetooth with my phone 
But not music 
The message I receive when trying to connect is "the device has been added, but failed to connect"
Thanks

closed 2017-09-15 by TxusLado

infirit, 2017-09-15:
>I have consulted the [Troubleshooting](https://github.com/blueman-project/blueman/wiki/Troubleshooting) page and done my best effort to follow.

Did you?

>blueman:
>BlueZ:
>Distribution:
>Desktop environment:

And only one was provided... 😢