blueman-project/blueman #773: Bluetooth pairing with cell phone as network adapter sometimes fails with "Connection Failed: uid 1000 has no permission to perform this operation"

msp3k, 2017-09-17:
blueman: 2.0.4-1ubuntu2
BlueZ: 5.37-0ubuntu5
Distribution: Ubuntu 16.04
Desktop environment: Mate

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

Connecting to a cell phone for use as a network adapter sometimes fails with "Connection Failed: uid 1000 has no permission to perform this operation".  Running blueman-manager as root is a workaround.

cschramm, 2017-09-23:
That message's coming from NetworkManager. `/usr/share/polkit-1/rules.d/60-network-manager.rules` should allow sudo and netdev users to modify system settings. Is your user (uid 1000) in one of those groups?

Predatux, 2017-10-05:
Hi, I have the same problem. Can you tell me what steps I need to take to fix it?
Thank you in advance

amalbuquerque, 2017-11-17:
Hey, I had the exact same problem on Ubuntu 17.10. The previous comment led me in the right direction. It was a matter of adding the current user to the `netdev` group:
```
sudo gpasswd -a $USER netdev
```

closed 2017-11-21 by cschramm

Gable, 2019-07-01:
This exact problem just occurred to me in Ubuntu Disco Dingo and my user id was already part of the sudo group.
I guessed that there must be another group that my userid needed to be a part of so I added my userid to the bluetooth group and that seems to have solved the problem.
But! The problem shouldn't happen in the first place. Anyone who can connect to a WiFi access point with NetworkManager should be able to pair to a bluetooth device.  The udev rules should allow that by default.
Also, in the interim, the error message should be improved from:
"Connection Failed: uid 1000 has no permission to perform this operation"
to something that indicates how to correct the problem, something like:
"userid needs to be added to the netdev group to perform this operation"

cschramm, 2019-07-03:
Hi @Gable 

> my user id was already part of the sudo group

If you're sure the user was effectively in the `sudo` group (i.e. you did not just add him after the last boot so that it might not have been known in the correct context) that's either due to some change in NetworkManager (probably a Debian / Ubuntu packaging bug then) or some issue with the rules (a packaging bug). If you're sure about it, you should file a report for that at https://launchpad.net/ubuntu/+source/network-manager/+bugs.

> Anyone who can connect to a WiFi access point with NetworkManager should be able to pair to a bluetooth device.

Regarding WiFi vs Bluetooth networking, NetworkManager uses a single `org.freedesktop.NetworkManager.settings.modify.system` polkit action in any case.

Regarding networking vs. Bluetooth, your distribution does not follow the concept you're expecting. The network-manager package installs polkit rules that allow networking tasks to the `sudo` and `netdev` groups while the bluez package installs D-Bus configuration that allows bluetoothd usage to the `root` user and the `bluetooth` and `lp` groups.

That concept makes sense to me, but if you really think networking and Bluetooth permissions should get aligned, https://launchpad.net/ubuntu/+source/network-manager/+bugs and / or https://launchpad.net/ubuntu/+source/bluez/+bugs is the correct place to ask for that.

> Also, in the interim, the error message should be improved from:
> "Connection Failed: uid 1000 has no permission to perform this operation"
> to something that indicates how to correct the problem, something like:
> "userid needs to be added to the netdev group to perform this operation"

That's really hard to achieve. The error message is from NetworkManager (https://gitlab.freedesktop.org/NetworkManager/NetworkManager/blob/bdc57c3d70941546fa0c18d4e3f014ab42e3bf21/src/nm-auth-utils.c#L550) and as the polkit rules are a distribution / admin thing, upstream has no idea of the fact that in your system your user would have to be added to the `netdev` group (`netdev` is a perfect example as it's a Debian thing and does not exist anywhere else). So the correct way to achieve what you're asking for would be that the Ubuntu maintainer overrides the upstream error message. Again, https://launchpad.net/ubuntu/+source/network-manager/+bugs if you really want to ask for that.