blueman-project/blueman #775: Question on rfcomm from @parkerlreed

infirit, 2017-09-26:
>I have a Bluetooth device (serial port) that I currently use a Python script in a loop to: check if the device is broadcasting (on), and auto rfcomm bind if it is.

>Question is: would it be possible to maybe implement this as a plugin for Blueman? Idea is basically, "When device is in range, run arbitrary command (or run an action like binding to an rfcomm port)

@parkerlreed, if you do it in a script I am fairly sure it can be done in a plugin. However bluemans plugins are not meant for users to install and maintain. It then boils down to is it something we are willing to maintain in blueman. Post the script in a gist and lets see what exactly you are doing.

ps. I personally have no serial/rfcomm use except for basic testing on my speakers. @cschramm I think does have some serial equipment but I am not sure how much he uses it. If we both don't use it it will be hard to know if we have not broken things along the way.

parkerlreed, 2017-09-27:
It's a little crude but here it is

````
#!/usr/bin/python2

import bluetooth
import time
from subprocess import call

print "In/Out Board"
btval = 3

while True:

    result = bluetooth.lookup_name('00:18:91:D8:FF:72', timeout=3)
    if (result != None):
	if (btval == 0 ):
            btval = 1
            time.sleep(5)
            call(["sudo", "rfcomm", "bind", "0", "00:18:91:D8:FF:72"])
            time.sleep(5)
            call(["/home/parker/bin/xgpsandfox"])
        else:
            print "Already connected"
    else:
        print "Disconnected"
        call(["sudo", "rfcomm", "release", "0"])
        btval = 0

    time.sleep(1)
````

My thought process was Blueman might already be doing some checks for availability which could be used to perform actions.

parkerlreed, 2017-09-27:
And here's me explaining it from a day or two ago lol https://www.youtube.com/watch?v=7WetQOOEBIY

infirit, 2017-09-27:
>And here's me explaining it from a day or two ago

"And btval is eeeuh..... its not connected" made me chuckle 😄.

We already have SerialManager that allows you to run custom commands when an rfcomm device is connected. See https://github.com/blueman-project/blueman/blob/master/blueman/plugins/applet/SerialManager.py.

>My thought process was Blueman might already be doing some checks for availability which could be used to perform actions.

Not automatically, discovery is only started when the user exp;licitly requests it. We currently do not have is a plugin that starts discovery every X seconds and then takes action when a registered device is found. I'll have to think about this a bit and see if this can be useful for others.

If you want to try yourself all the functions that are run for plugins when certain event happen are here: https://github.com/blueman-project/blueman/blob/master/blueman/plugins/AppletPlugin.py#L57

What you will want to do is ``start_discovery`` on the Adapter class and then handle connecting in ``on_device_created``.

parkerlreed, 2017-09-27:
>We already have SerialManager that allows you to run custom commands when an rfcomm device is connected

That much I already automated in udev, but great to know it's there :D

I'll have to take a look at this in a little bit. I know just enough Python to be dangerous, so I can't promise any results ;)

parkerlreed, 2017-10-03:
Ok so after testing the rfcomm_channel branch (which works well by the way :smile:) had some new thoughts.

Maybe just a general auto-connect feature? Bluetooth speakers/mice already do this (through the agent registration if I recall). Really wish I knew more Python to be able to dig into this :)

parkerlreed, 2017-10-03:
Yeah. In my head I was imagining maybe a dialog with the known devices with a checkmark to activate auto-connecting just for that device. (Or maybe make it part of the existing Setup process?)

cschramm, 2017-10-03:
If it's just per-device, it should be similar to the "trusted" attribute, but I don't think the logic can be that simple. It's rather a per-service setting. Think of a phone that provides audio and network services. You might want to auto-connect the audio services when it becomes available but not the network service and vice versa.

closed 2019-04-27 by stale[bot]