blueman-project/blueman #776: Fedora 26 Bluetooth enables itself when disabled previously before laptop close  / sleep.

dippedinsauce, 2017-09-28:
blueman:
BlueZ:
Distribution: fedora 26 
Desktop environment: xfce

This is validated through the gui/network manager applet.  Please let me know what dumps you need and how to generate.  I keep bluetooth off due to security issues an reduction of attack surface.  It is a bit concerning with the active flaws recently known that when I come out of sleep, bluetooth is enabled and I have to manually turn it off every time.  If this is the wrong group and is a fedora bug, please lead me to where the bug can be filed appropriately.  Thanks!

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

> ℹ Please use some service like [Pastebin](http://pastebin.com/) or [GitHub Gist](https://gist.github.com/) to post your logs and keep the thread clean and readable. Make sure to describe the exact steps you took when producing it.

infirit, 2017-09-28:
Have a look in the plugin dialog in PowerMmanage under the configuration button. There should be a auto power on feature which is enabled by default.

infirit, 2017-10-19:
No response in 3 weeks, closing. Feel free to comment if blueman is still powering the adapter.

closed 2017-10-19 by infirit